MEN-Mikro-Elektronik/13MD05-90 #133: jpe-dev branch does not compile due to MENT_XSTR and BBIS_Entry

RBachmannMEN, 2019-07-10:
For the last few days, the jpe-dev branch does not build on my G22 with Ubuntu. The issue seems to relate to MENT_XSTR and also BBIS_Entry.
I know that jpe-dev is a development branch, however, I need to use it for my work and it would be nice if you could fix it.

The exact errors are:

```
/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.c:129:33: error: implicit declaration of function ‘MENT_XSTR’; did you mean ‘ENOSTR’? [-Werror=implicit-function-declaration]
 static const char IdentString[]=MENT_XSTR(MAK_REVISION);
                                 ^~~~~~~~~
                                 ENOSTR
<command-line>:0:14: error: invalid suffix "MD05" on integer constant
/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.c:129:43: note: in expansion of macro ‘MAK_REVISION’
 static const char IdentString[]=MENT_XSTR(MAK_REVISION);
                                           ^~~~~~~~~~~~
<command-line>:0:21: error: invalid suffix "_02_00" on integer constant
/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.c:129:43: note: in expansion of macro ‘MAK_REVISION’
 static const char IdentString[]=MENT_XSTR(MAK_REVISION);
                                           ^~~~~~~~~~~~
<command-line>:0:34: error: ‘g1e4ac0a’ undeclared here (not in a function)
/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.c:129:43: note: in expansion of macro ‘MAK_REVISION’
 static const char IdentString[]=MENT_XSTR(MAK_REVISION);
                                           ^~~~~~~~~~~~
<command-line>:0:43: error: ‘dirty_2019’ undeclared here (not in a function)
/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.c:129:43: note: in expansion of macro ‘MAK_REVISION’
 static const char IdentString[]=MENT_XSTR(MAK_REVISION);
                                           ^~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:323: recipe for target '/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.o' failed
make[4]: *** [/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2/bb_smb2.o] Error 1
scripts/Makefile.build:579: recipe for target '/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2' failed
make[3]: *** [/home/rafael/13MD05-90/OBJ/nodbg/men_bb_smb2] Error 2
Makefile:1534: recipe for target '_module_/home/rafael/13MD05-90/OBJ' failed
make[2]: *** [_module_/home/rafael/13MD05-90/OBJ] Error 2
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:498: recipe for target 'callkernelbuild' failed
make[1]: *** [callkernelbuild] Error 2
/opt/menlinux/BUILD/MDIS/TPL/rules.mak:400: recipe for target 'nodbg' failed
make: *** [nodbg] Error 2
```

KonradJPE, 2019-07-10:
Could you please provide the Makefile, as well as gcc and kernel version?

RBachmannMEN, 2019-07-10:
Here is my Makefile (with .txt type because github does not support Makefile as plain filename: [Makefile.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3377320/Makefile.txt)

GCC: `gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)`
kernel: `4.18.0-24-generic #25~18.04.1-Ubuntu SMP`

mlassota, 2019-07-10:
There should be these lines in include file
*vim* **/opt/menlinux/INCLUDE/COM/MEN/men_typs.h** *+169*
~~~~
/** stringize macros */
#define MENT_XSTR_SFX(s,p) MENT_STR(s) " " MENT_STR(p)
#define MENT_XSTR(s) MENT_STR(s)
#define MENT_STR(s) #s
~~~~
If these lines are missing, please make an installation again (**sudo ./INSTALL.sh**)
I get this error when I comment out *#define MENT_XSTR(s) MENT_STR(s)* line.
I hope this will help.

RBachmannMEN, 2019-07-10:
These lines were indeed missing, however when I clone again all is ok.

This problem occurred because I checked out master (with --recursive), then checked out jpe-dev. 

It works now, thank you for your help!

closed 2019-07-10 by KonradJPE