MEN-Mikro-Elektronik/13MD05-90 #142: 13Z025-90: Add module parameter to disable the automatic type determination of the UART's

AndreasGeissler, 2019-08-07:
Hello,

We have found that under certain circumstances, the Linux UART serial driver 8250 sends data through the UART to determine the UART type. We have a board that has problems with these bytes sent.

Therefore we need an option for the 13Z025-90 driver to provide to possibility to disable this autoscan feature of the driver:

An addition module parameter shall be added:
 module_param( fixed_type, char, 0 );

if this parameter is set to true (a value different to 0), than the flag for the UART port to force a fixed type shall be added:
 men_uart_port.port.flags |= UPF_FIXED_TYPE;

The fixed type shall be than set to:
 men_uart_port.port.type = PORT_16550A;

Regards,
 Andreas

KonradJPE, 2019-08-08:
fixed_type parameter added. Can be used for both z025 and z125. 
Please check if this solves the problem. I will close the issue if is working as expected.

dpfeuffer, 2019-10-02:
Andreas and I reviewed your modification in men_z25_serial.c and it seems ok. We cannot perform a functional test yet, but we assume it will work ;-)
The only thing I'm wondering about is why you added fixed_type parameter as string (charp) instead of char (byte). But it shouldn't matter.

closed 2019-10-02 by dpfeuffer