MEN-Mikro-Elektronik/13MD05-90 #144: Improvements of scan_system.sh

KonradJPE, 2019-08-09:
Due to significant increase of code in scan_system.sh, we would like to divide scan_system script into several sources.
1. Move Xml parser function into separte source file,
2. Move User interaction functions, and those that can be used regardless of scanning into separate source file,
3. Make bash source consistent, code clean up (like intendation).

dpfeuffer, 2019-08-09:
I searched the web if it is common to split a large shell script into smaller peaces and found this discussion about it:
https://unix.stackexchange.com/questions/74994/is-it-common-to-split-larger-script-into-multiple-scripts-and-source-them-in-the
Because it is easier for the user to have just one script file, please consider this approach:
https://github.com/zinovyev/bash-project

Currently, the MDIS User Manual contains some hints to the one scan_system.sh script file. If more than one file is used, we have to modify some descriptions, e.g.
*In cross compilation environments, you could copy scan_system.sh to your target system, perform  the  scan  manually  and  then  copy  back  the  generated  MDIS  configuration  to your host system.*
But this shouldn't be an obstacle not to split the quite large scan_system.sh script for better maintenance into some smaller peaces. I understand your reasons well.

mlassota, 2019-08-09:
Yep, there's always pros and cons.
The pro is it adds more readability and makes it easy to maintain (at least it should), and the con is it's bad for the end user.
Anyhow It's up to you to decide.

dpfeuffer, 2019-08-09:
As programmer I vote for more readability and easier maintenance ;-)
What do you think about this approach? https://github.com/zinovyev/bash-project

mlassota, 2020-01-15:
> What do you think about this approach? https://github.com/zinovyev/bash-project

It looks cool, however source code amalgamation is not good when it comes to debugging.
Suppose we went this way. If an error (syntax) or bug (logic) occurs, we will have to find it first in this amalgamated script, propably fix it there, then find this buggy part in the source script and then make the final fix.

dpfeuffer, 2020-01-20:
Ok, then we better keep it simple with this approach: Split scan_system.sh in several .sh files and group all the files together by storing it in a subdir (e.g. scan_system).
So it is easy to describe ".. copy scan_system dir to your target system .." and we have no overhead to generate a scan_system.sh from other sources. However the call path in mdiswiz and INSTALL.sh must be adapted (or a symbolic link to scan_system/scan_system.sh provided).

closed 2020-05-27 by KonradJPE