MEN-Mikro-Elektronik/13MD05-90 #145: mdiswiz: add Note column to module selection dialogue for EOL/EOS

dpfeuffer, 2019-08-12:
Prior information to new requirement in 13MD05-90_SwArchSpec:
Please add a Note column to all module selection dialogues to indicate EOL/EOS of individual modules (CPU boards, carriers, IP cores, ...).
Suggestion:
[image]

We will provide you a list which modules must be marked as EOL or EOS.

dpfeuffer, 2019-08-12:
After a discussion with Florian, we want to use the term EOL for EOL and EOS products because the difference between EOL/EOS may be confusing for the customer.
So please reduce it to EOL in the Note column and to the description:
EOL (End of Life):
No longer maintained. Use on your own risk.

KonradJPE, 2019-08-28:
I have additional question regarding some of the components that are defined in 13MD05-90_SwArchSpec EOS/EOL list. This is due inconsistency between the names used in mdiswiz and full component names. Some of the components are not visible in mdiswiz at all. 
Please see attached file. 
[unknownComponents.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/3550272/unknownComponents.log)

JSierl, 2019-08-30:
You are right with your mapping inside the unknownComponents.log
Sometimes (as with F19P and F19C) they have the same basis and were therefore mentioned as one in mdiswiz. 

Regarding the products missing in mdiswis but listed as EOL/EOS in the SwArchSpec: Just leave it be. Don't add them only to mark them as deprecated.

dpfeuffer, 2019-09-16:
During my test of https://github.com/MEN-Mikro-Elektronik/13MD05-90/commit/d97924eb3f4619c1e8bb2650e96ef55ac4d2b162 for #147 I missed the description:
```
EOS (End of Support):
No longer maintained. See supported_components.md for details.
```
In red, see my comment from 12 Aug above.

Was the provided binary created before this issue was closed or is the red description now added?

KonradJPE, 2019-09-23:
> During my test of d97924e for #147 I missed the description:

You are right, EOS note was missing while creating new project in CPU section. In other windows description was present. 

I have added updated mdiswiz binary into repository.

dpfeuffer, 2019-09-24:
Ok, then I close this issue.

dpfeuffer, 2019-10-25:
I reopen this issue, because of the additional requirement to still support:
-	04M035N
-	04M072-
-	04M077N
Please remove EOS note for this M-Modules.
I updated lists of EOS/EOL components (SA_0450, SA_0460) in 13MD05-90_SwArchSpec but the spec is not yet released.

dpfeuffer, 2019-10-25:
The 16Z072_OWB IP core is specified as EOS (see MEN_13MD05-90_SA_0450). Please mark it as EOS.
Unfortunately, the belonging 13Z072_OWB driver is not included in 13MD05-90 as assumed. I will open another issue for the integration.

KonradJPE, 2019-10-28:
> I reopen this issue, because of the additional requirement to still support:
>     04M035N
>     04M072-
>     04M077N
> Please remove EOS note for this M-Modules.

EOS note removed from below M-Modules:
- M35
- M72
- M72_PRE
- M77

> The 16Z072_OWB IP core is specified as EOS

Added EOS flag for 16Z072_OWB