MEN-Mikro-Elektronik/13MD05-90 #147: mdiswiz: Headline font not readable at XFCE desktop

dpfeuffer, 2019-08-14:
Linux vm-manjaro 5.2.2-1-MANJARO (Xfce 4.13, 4.14)
Used MDIS Wizard: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/112#issuecomment-513194487

1. The headline of the mdiswiz startup dialog is not readable under XFCE.
2. The other fonts are readable but not really good displayed under XFCE.

[image]

Please investigate if the Qt font could be changed to solve 1. and 2..
The fonts should be clear readable under this desktop environments:
- GNOME
- XFCE
- KDE Plasma
- LXDE

If this makes to much effort, please change the headline of the mdiswiz startup dialog to be readyble at all (to solve 1.). 

Currently there is no requirement which desktop environments shall be supported.
I don't want to define a new requirement in the SwArchSpec at the moment to minimize the testing effort.

Additional remark:
mdiswiz binary of 13MD05-90_02_00 leads to just a segfault but doesn't start at all. But yes, we have no requirement to support manjaro ;-).

mlassota, 2019-08-29:
>1. The headline of the mdiswiz startup dialog is not readable under XFCE.
>2. The other fonts are readable but not really good displayed under XFCE.

Fixed. Please check the binary from commit d97924eb3f4619c1e8bb2650e96ef55ac4d2b162

Actually I did not change anything in source code. It's the libraries mdiswiz is linked with does matter. Had to test a lot of different versions with a lot of configuration parameters.

closed 2019-08-29 by mlassota

dpfeuffer, 2019-09-16:
>Fixed. Please check the binary from commit d97924e

I tested the binary under Linux vm-manjaro 5.2.2-1-MANJARO (Xfce 4.13, 4.14) with success.
1. and 2. fixed and the binary is running under Manjaro. Great, thanks!