MEN-Mikro-Elektronik/13MD05-90 #149: Lack of support in mdiswizard for components mentioned in 13MD05-90_02_01 requirements

KonradJPE, 2019-08-28:
mdiswizard does not support below components:
- CPU board 01A025,
- MEN Display Computer 09DC19,
- MEN Display Computer 09DC20,
- M-Module M47,
- M-Module M57,
- M-Module M66,
-M-Module M199.

Components are not visible in module selection dialogue.

JSierl, 2019-08-30:
That's absolutely right, they are not yet supported by mdiswiz. It's your task to add them now :-)

dpfeuffer, 2019-09-16:
>CPU board 01A025,
> MEN Display Computer 09DC19,
> MEN Display Computer 09DC20,

CPU models must be added to mdiswiz.

> M-Module M47 (lack of submodule in 13MD05-90)
> M-Module M57 (lack of repository),
> M-Module M199 (lack of repository).

M47 and M199 drivers must be added as submodules to 13MD05-90 master repo and are available:
- https://github.com/MEN-Mikro-Elektronik/13M047-06
- https://github.com/MEN-Mikro-Elektronik/13M199-06

13M057-06 is a pay article that contains third-party sources and must not be added to 13MD05-90 master repo. The sources are available from our support department on request. See https://www.men.de/software/13m057-06/.

KonradJPE, 2019-09-24:
> M47 and M199 drivers must be added as submodules to 13MD05-90 master repo and are available:
> 
>     https://github.com/MEN-Mikro-Elektronik/13M047-06
>     https://github.com/MEN-Mikro-Elektronik/13M199-06
> 

Added

KonradJPE, 2019-09-25:
@dpfeuffer Question regarding DC19/DC20.

Should the configuration be exactly the same (except from the name) as for BL/BC51 which is also CPU Board SC31?

dpfeuffer, 2019-09-26:
Yes.
I checked the DC19/DC20 specification and see no difference to the BL/BC51. All CPU units use the SC31 (with FPGA and SMBus).
I would prefer to add one additional model "DC19/DC20" instead of extending the existing "Bx51x" model to "Bx51x/DC19/DC20". Then we have still the possibility to change the parameters if I overlook anything ;-).

KonradJPE, 2019-09-30:
>     CPU board 01A025,
>     MEN Display Computer 09DC19,
>     MEN Display Computer 09DC20,
> 
> CPU models must be added to mdiswiz.

Added.

KonradJPE, 2019-10-01:
All components have been added into mdiswiz / mdis repository.

closed 2019-10-01 by KonradJPE