MEN-Mikro-Elektronik/13MD05-90 #150: men_lx_z77 compilation error on kernel 5.2.0 and newer

mlassota, 2019-09-23:
~~~
men_16z077_eth.c:1190:3: error: ‘struct ethtool_ops’ has no member named ‘get_settings’; did you mean ‘get_strings’?
  .get_settings = z77_ethtool_get_settings,
   ^~~~~~~~~~~~
   get_strings
men_16z077_eth.c:1190:18: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .get_settings = z77_ethtool_get_settings,
                  ^~~~~~~~~~~~~~~~~~~~~~~~
men_16z077_eth.c:1190:18: note: (near initialization for ‘z77_ethtool_ops.get_regs_len’)
men_16z077_eth.c:1191:3: error: ‘struct ethtool_ops’ has no member named ‘set_settings’; did you mean ‘get_strings’?
  .set_settings = z77_ethtool_set_settings,
   ^~~~~~~~~~~~
   get_strings
men_16z077_eth.c:1191:18: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .set_settings = z77_ethtool_set_settings,
                  ^~~~~~~~~~~~~~~~~~~~~~~~
men_16z077_eth.c:1191:18: note: (near initialization for ‘z77_ethtool_ops.get_regs’)
men_16z077_eth.c: In function ‘z77_bd_setup’:
men_16z077_eth.c:1259:16: error: implicit declaration of function ‘dma_zalloc_coherent’; did you mean ‘dma_alloc_coherent’? [-Werror=implicit-function-declaration]
   memVirtDma = dma_zalloc_coherent(&pcd->dev, Z77_ETHBUF_SIZE,
                ^~~~~~~~~~~~~~~~~~~
                dma_alloc_coherent
men_16z077_eth.c:1259:14: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
   memVirtDma = dma_zalloc_coherent(&pcd->dev, Z77_ETHBUF_SIZE,
              ^
~~~

In kernel 5.0.0 dma_zalloc_coherent has been dropped. dma_alloc_coherent should be used instead.
In kernel 5.2.0 get_settings and set_settings callbacks have been dropped. get_link_ksettings and set_link_ksettings should be implemented instead.

mlassota, 2019-09-23:
@dpfeuffer 
Dieter, do you want us to fix this or you want to handle this at MEN?

dpfeuffer, 2019-09-24:
Please fix it and verify the functionality (e.g. with your JPE_13MD05-90_ST_xxxx_F023P_Interface_Test).
The driver is required according MEN_13MD05-90_SA_1460 [FUNC] IP Core Z087_ETH.

mlassota, 2019-09-27:
Fixed and tested.

closed 2019-09-27 by mlassota