MEN-Mikro-Elektronik/13MD05-90 #151: Repo branches cleanup

dpfeuffer, 2019-10-04:
Evaluate existing branches and perform/assign branch cleanup.
Define how to deal with new temporary branches.

dpfeuffer, 2019-12-13:
1) I deleted some unused branches.
2) Feature/fix branches, shall be deleted after performed merges of pull requests from the person who did the merge. See also https://github.com/MEN-Mikro-Elektronik/13MD05-90/wiki/CM_Guideline “Workflow C - Outsourced projects”.
3) The following stale branches are left and must be evaluated later:
  - from Thomas Schnürer for MSI support in 13MD05-90:
    - https://github.com/MEN-Mikro-Elektronik/13MD05-90/tree/thomas-msi-support
    - https://github.com/MEN-Mikro-Elektronik/13MD05-90/tree/thomas-msi-support-2
  - from Harry about M77 and VME (PLDZ002):
    - https://github.com/MEN-Mikro-Elektronik/13MD05-90/tree/harry-m77-fix
    - https://github.com/MEN-Mikro-Elektronik/13MD05-90/tree/harry_fix_pldz002_module_removal

dpfeuffer, 2020-06-19:
We will leave both MSI related branches from Thomas Schnürer alive without merging it into master now.

harry_fix_pldz002_module_removal is already merged.

Please evaluate if /harry-m77-fix should be merged into master.

mlassota, 2020-07-15:
> Please evaluate if /harry-m77-fix should be merged into master.

I merged this into jpe-dev branch. It also needed a fix regarding #205

closed 2020-07-15 by mlassota