MEN-Mikro-Elektronik/13MD05-90 #154: LIBSRC subfolders: documentation files missing

dpfeuffer, 2019-10-10:
Please check library documentation (13MD05-90/MDISforLinux/LIBSRC and submodules).
Some documentation files may be missing and doxygen documentation may be incomplete (links doesn't work).

Example OSS:
- COM/oss_specification.c missing
- DOC/html/index.html: Link to the to the *OSS Common Specification* doesn't work 

I will provide COM/oss_specification.c via merge request.
Please check other library documentation for consistency.
See also MEN_13MD05-90_SA_0380	[DES] Documentation.

dpfeuffer, 2019-10-18:
Long time ago, each low-level driver package required at least a *_min.dsc and *_max.dsc descriptor file. At this time we hadn't neither .xml package description files nor the MDIS Wizard.
These .dsc files are no longer necessary, because the configuration information is stored in the .xml files settinglist section (for low-level drivers) and in the MDIS Wizard (for BBIS drivers).
For compatibility reasons, the *_min.dsc and *_max.dsc files were still present in the driver packages, but with the MCVS to Git migration (or later, don't know) the .dsc files were removed.
Unfortunately, the mentioning of these files (links, etc) were not considered at this time :-(
Please, remove all mentioning of the  *_min.dsc and *_max.dsc files. I found quite a lot matching lines in the current jpe-dev branch:
```
grep -Hrn "m..\.dsc" | wc --lines
258
```
I will add a requirement in the SwArchSpec chapter "4.6.2 MCVS Cleanup" for the 13MD05-90_02_02 release and a belonging test case to the MDIS acceptance test for 13MD05-90_02_01 delivery. So please consider this task now (add it to your time estimation for Johannes).
Thanks for your question/hint!

mlassota, 2019-10-22:
Other library documentation looks OK.

closed 2019-10-22 by mlassota