MEN-Mikro-Elektronik/13MD05-90 #156: INSTALL script root privileges

AndreasGeissler, 2019-10-18:
Hi,

Currently, it seems that the INSTALL script requires root privileges in all cases. This is a problem for Yocto projects that use MDIS. For non-root installation, we need an additional option or a check to see if root privileges are really required.

Regards,
Andreas

JSierl, 2019-10-18:
This is probably an additional detail on the already existing requirement to support yocto. So actually it isn't a new issue but should be kept as a reminder.

KonradJPE, 2019-10-21:
Hi Andreas

> Currently, it seems that the INSTALL script requires root privileges in all cases. This is a problem for Yocto projects that use MDIS. For non-root installation, we need an additional option or a check to see if root privileges are really required.

I am not sure if I understand it correctly. What kind of problem appears on Yocto distro? 

Root piviliges are necessary in several situations:
- creating dir/files in opt/ directory
- detecting men hardware (commands like i2cdump) - in default scan_system scripts starts automatically while running INSTALL script. 

We can change the script to prior check if current user has proper rights to make changes in the default 'root' directories. Is it the change you would like to add?

AndreasGeissler, 2019-10-21:
Hi Konrad,

I'm sorry that I did not describe this problem clearly. It's just a problem when creating MDIS with Yocto, but not on an already running system..

The problem is, that it is a very bad idea to use root privileges for Yocto building process. There is already a good feature for Yocto inside the INSTALL script which allows only to install MDIS without the scan feature and without asking interactively questions ("--install-only"). Would it be possible that in cases where no root privileges are required, the check for the root will be disabled?

For example: When installing MDIS ony to a specific path where no root privileges are needed?

Regards,
Andreas

KonradJPE, 2019-10-22:
> For example: When installing MDIS only to a specific path where no root privileges are needed?

Install script should now work without root privileges in situation as described below. 
Root privileges check is disabled only when "--install-only" and proper "--path=..." option is used.
Path has to be accessible without root privileges.

closed 2019-10-22 by KonradJPE