MEN-Mikro-Elektronik/13MD05-90 #159: mdiswiz: creation of fontconfig dir causes problems

dpfeuffer, 2019-11-05:
Linux vm-manjaro 5.3.6-1-MANJARO #1 SMP PREEMPT Sat Oct 12 09:30:05 UTC 2019 x86_64 GNU/Linux
[dpfeuffer@vm-manjaro 13MD05-90]$ git describe --all --long 
heads/jpe-dev-0-gc502e8e

In the curren jpe-dev branch, 13MD05-90/MDISforLinux/BIN/mdiswiz creates an unwanted fontconfig dir in the working dir that could cause the following problem:

```
[dpfeuffer@vm-manjaro proj]$ mkdir m11
[dpfeuffer@vm-manjaro proj]$ cd m11
[dpfeuffer@vm-manjaro m11]$ ~/work/repos/13MD05-90/MDISforLinux/BIN/mdiswiz & 
```
selection of MDIS project dir "/home/dpfeuffer/work/mdis/proj/m11/" causes error:
```
MDIS_PROJ_DIR: "/home/dpfeuffer/work/mdis/proj/m11/fontconfig/" is an existing, non-empty directory!
```

Please try to prevent creation of fontconfig dir or create it under another location (e.g. /tmp/mdiswiz).

mlassota, 2019-11-07:
It's the *fontconfig* causing this (*mdiswiz* is statically-linked with *fontconfig* library). *fontconfig* reads configuration file */etc/fonts/fonts.conf* on your computer, that contains font cache directory list, e.g. 
```
<!-- Font cache directory list -->
        <cachedir>/var/cache/fontconfig</cachedir>
        <cachedir prefix="xdg">fontconfig</cachedir>
        <!-- the following element will be removed in the future -->
        <cachedir>~/.fontconfig</cachedir>
```      
This configuration file takes precedence over built-in configuration.
When *fontconfig* wants to write font cache it selects the first cache directory from the list it can write and since ```<cachedir>/var/cache/fontconfig</cachedir>``` is not accessible without root privileges it selects  ```<cachedir prefix="xdg">fontconfig</cachedir>``` which basically means ```$(pwd)/fontconfig``` (i.e. 'current working directory'/fontconfig).
I can change the path where *fontconfig* searches for configuration, but then this error will appear on output:
```
Fontconfig error: Cannot load default config file
```
and font selection will be broken, however font cache will be written in the right directory ```/tmp/mdiswiz/fontconfig``` (from built-in configuration).
This bug occurs since they've decided to remove  ```<cachedir>~/.fontconfig</cachedir>``` in the future. ```<cachedir prefix="xdg">fontconfig</cachedir>``` has jumped in this place.

dpfeuffer, 2019-11-12:
If I understand it right, we should keep it as it is, otherwise, "font selection will be broken". I wasn't aware, that the problem exists only when running mdiswiz as normal user.
We recommend in the MDIS user manual to run mdiswiz as root:
"As the system scan and the kernel module installation (performable by mdiswiz) requires root privileges, it is recommended to always run scan_system.sh and mdiswiz as root."
I can add a remark in the user manual about this "known problem".
If you agree with me, please close this issue without changing mdiswiz.

closed 2019-11-12 by mlassota