MEN-Mikro-Elektronik/13MD05-90 #160: add fpga_load_ppc binary

dpfeuffer, 2019-11-13:
Build and add fpga_load_ppc binary to https://github.com/MEN-Mikro-Elektronik/13MD05-90/tree/jpe-dev/MDISforLinux/BIN.

And while you're at it: 
Please update PPC binaries of hwbug in https://github.com/MEN-Mikro-Elektronik/hwbug/tree/jpe-dev

mlassota, 2020-01-09:
@dpfeuffer 
I've managed to compile fpga_load for PowerPC (using QEMU and Debian for PowerPC), however I have no CPU to test if it is able to display Chameleon Table. Could you please check if it is running correctly?

[fpga_load.tar.gz](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4041169/fpga_load.tar.gz)

gvarlet, 2020-02-05:
I've tested successfully the fpga_load file made from Marcin (thanks!) :+1:  :

root@generic-powerpc-e500v2:~# fpga_load -s

Nr.| dom|bus|dev|fun| Ven ID | Dev ID | SubVen ID |
  0|  0   0   0   0   0x1957   0x0111    0x0000
  1|  0   2   0   0   0x1957   0x0111    0x0000
  2|  0   4   0   0   0x1957   0x0111    0x0000
  3|  0   5   0   0   0x1a88   0x4d45    0x00ab
root@generic-powerpc-e500v2:~# fpga_load 0x1a88 0x4d45 0x00ab 0 -t

BARs of FPGA PCI device 05:0.0:
  BAR0: 0xf0000000; size: 0x00000000, mapType: MEM;
  BAR1: 0x80000000; size: 0x00000000, mapType: MEM;
  BAR2: 0x00000000; size: 0x00000000, mapType: unused;
  BAR3: 0x00000000; size: 0x00000000, mapType: unused;
  BAR4: 0x00000000; size: 0x00000000, mapType: unused;
  BAR5: 0x00000000; size: 0x00000000, mapType: unused;

Information about the Chameleon FPGA:
FPGA File='A021-00ic001' table model=0x41('A') Revision 18.1 Magic 0xABCE
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x0018 16Z024_SRAM                0    0   1  16  63   0 0x00000000 0xf0000000
  1 0x007f 16Z127_GPIO                0    0   0   3   0   0 0x00000200 0xf0000200
  2 0x007e 16Z126_SERFLASH            0    0   0   8  63   0 0x00000220 0xf0000220
  3 0x004c 16Z076_QSPI                0    0   0   8   3   0 0x00002000 0xf0002000
  4 0x0042 16Z066_MMOD                1    0   1  13   1   0 0x00001400 0xf0001400
  5 0x0042 16Z066_MMOD                1    0   2   0  63   0 0x01000000 0xf1000000
  6 0x0042 16Z066_MMOD                2    1   1  13   1   0 0x00001800 0xf0001800
  7 0x0042 16Z066_MMOD                2    1   2   0  63   0 0x02000000 0xf2000000
  8 0x0042 16Z066_MMOD                3    2   1  13   1   0 0x00001c00 0xf0001c00
  9 0x0042 16Z066_MMOD                3    2   2   0  63   0 0x03000000 0xf3000000
 10 0x0002 16Z002_VME                 4    0   3   0  63   0 0x00010000 0xf0010000
 11 0x0002 16Z002_VME                 4    0   4   0  63   0 0x00020000 0xf0020000
 12 0x0002 16Z002_VME                 4    0   5   0  63   0 0x00400000 0xf0400000
 13 0x0002 16Z002_VME                 4    0   1   9   2   0 0x00800000 0xf0800000
 14 0x0002 16Z002_VME                 4    0   6   0  63   0 0x04000000 0xf4000000
 15 0x0002 16Z002_VME                 4    0   7   0  63   0 0x05000000 0xf5000000
 16 0x0002 16Z002_VME                 4    0   8   0  63   1 0x00000000 0x80000000

 Current FPGA file/usage status: fallback image active, no configuration error occurred.

root@generic-powerpc-e500v2:~# fpga_load 0x1a88 0x4d45 0x00ab 0 -z -u ./
.ash_history         01A021-00_01_28.bin
.bashrc              wdt
root@generic-powerpc-e500v2:~# fpga_load 0x1a88 0x4d45 0x00ab 0 -z -u ./01A021-0
0_01_28.bin 0
Update OK
root@generic-powerpc-e500v2:~#

Power Cycle...

root@generic-powerpc-e500v2:~# fpga_load 0x1a88 0x4d45 0x00ab 0 -t

BARs of FPGA PCI device 05:0.0:
  BAR0: 0xf0000000; size: 0x00000000, mapType: MEM;
  BAR1: 0x80000000; size: 0x00000000, mapType: MEM;
  BAR2: 0x00000000; size: 0x00000000, mapType: unused;
  BAR3: 0x00000000; size: 0x00000000, mapType: unused;
  BAR4: 0x00000000; size: 0x00000000, mapType: unused;
  BAR5: 0x00000000; size: 0x00000000, mapType: unused;

Information about the Chameleon FPGA:
FPGA File='     A021-00' table model=0x41('A') Revision 28.0 Magic 0xABCE
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x0018 16Z024_SRAM                0    0   1  16  63   0 0x00000000 0xf0000000
  1 0x007f 16Z127_GPIO                0    0   0   8   0   0 0x00000200 0xf0000200
  2 0x007e 16Z126_SERFLASH            0    0   0  13  63   0 0x00000220 0xf0000220
  3 0x004c 16Z076_QSPI                0    0   0  13   3   0 0x00002000 0xf0002000
  4 0x0042 16Z066_MMOD                1    0   1  13   1   0 0x00001400 0xf0001400
  5 0x0042 16Z066_MMOD                1    0   2   0  63   0 0x01000000 0xf1000000
  6 0x0042 16Z066_MMOD                2    1   1  13   1   0 0x00001800 0xf0001800
  7 0x0042 16Z066_MMOD                2    1   2   0  63   0 0x02000000 0xf2000000
  8 0x0042 16Z066_MMOD                3    2   1  13   1   0 0x00001c00 0xf0001c00
  9 0x0042 16Z066_MMOD                3    2   2   0  63   0 0x03000000 0xf3000000
 10 0x0002 16Z002_VME                 4    0   3   0  63   0 0x00010000 0xf0010000
 11 0x0002 16Z002_VME                 4    0   4   0  63   0 0x00020000 0xf0020000
 12 0x0002 16Z002_VME                 4    0   5   0  63   0 0x00400000 0xf0400000
 13 0x0002 16Z002_VME                 4    0   2  16   2   0 0x00800000 0xf0800000
 14 0x0002 16Z002_VME                 4    0   6   0  63   0 0x04000000 0xf4000000
 15 0x0002 16Z002_VME                 4    0   7   0  63   0 0x05000000 0xf5000000
 16 0x0002 16Z002_VME                 4    0   8   0  63   1 0x00000000 0x80000000
 17 0x0002 16Z002_VME                 4    0   9   0  63   2 0x00000000 (nil)

 Current FPGA file/usage status: fallback image active, no configuration error occurred.

root@generic-powerpc-e500v2:~#

mlassota, 2020-02-06:
fpga_load_ppc added.

closed 2020-02-06 by mlassota