MEN-Mikro-Elektronik/13MD05-90 #161: F223 detection scan_system.sh flaw.

KonradJPE, 2019-11-18:
In some board configuration F223 is not detected by scan_system script. 
More description details will follow.

mlassota, 2019-11-18:
# Configuration that F223 is not detected by scan_script.sh
```
 standard cPCI slots    cPCI Serial slots
  
cpu,1 cpu,1 cpu,1       cpu,2 cpu,3 cpu,4 cpu,5
  4     3     2    SYS    2     3     4     5
 ---   ---   ---   ---
 | |   | |   | |   | | 
 | |   | |   | |   | | 
 | |   | |   | |   | | 
 | |   | |   | |   | |
 | |   | |   | |   | |   ----  ----  ----  ----
 | |   | |   | |   | |   |  |  |  |  |  |  |  |
 ---   ---   ---   ---   ----  ----  ----  ----
        ^     ^     ^            ^
        |     |     |            |
        |     |     |            +---G204+M82
        |     |     +---F26L
        |     +---F205+M11+M43
        +---F223
```

<details>
<summary>Output of <code>$ lspci -v -nn</code></summary>
<p>

```
00:00.0 Host bridge [0600]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Host Bridge [8086:5af0] (rev 0b)
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Flags: bus master, fast devsel, latency 0

00:02.0 VGA compatible controller [0300]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Integrated Graphics Controller [8086:5a84] (rev 0b) (prog-if 00 [VGA controller])
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation Device [8086:2212]
	Flags: bus master, fast devsel, latency 0, IRQ 139
	Memory at 99000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 80000000 (64-bit, prefetchable) [size=256M]
	I/O ports at f000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: i915
	Kernel modules: i915

00:0e.0 Multimedia audio controller [0401]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Audio Cluster [8086:5a98] (rev 0b)
	DeviceName: Onboard - Sound
	Subsystem: Intel Corporation Device [8086:7270]
	Flags: bus master, fast devsel, latency 0, IRQ 25
	Memory at 9ae10000 (64-bit, non-prefetchable) [size=16K]
	Memory at 9a800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_skl

00:0f.0 Communication controller [0780]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Trusted Execution Engine [8086:5a9a] (rev 0b)
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Flags: bus master, fast devsel, latency 0, IRQ 160
	Memory at 9ae35000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: mei_me
	Kernel modules: mei_me

00:12.0 SATA controller [0106]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SATA AHCI Controller [8086:5ae3] (rev 0b) (prog-if 01 [AHCI 1.0])
	DeviceName: Onboard - SATA
	Subsystem: Intel Corporation Device [8086:7270]
	Flags: bus master, 66MHz, medium devsel, latency 0, IRQ 137
	Memory at 9ae14000 (32-bit, non-prefetchable) [size=8K]
	Memory at 9ae32000 (32-bit, non-prefetchable) [size=256]
	I/O ports at f090 [size=8]
[... 924 lines cut ...]
	Kernel modules: shpchp

15:01.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 134
	Bus: primary=15, secondary=16, subordinate=16, sec-latency=0
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp

15:02.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 135
	Bus: primary=15, secondary=17, subordinate=17, sec-latency=0
	Capabilities: <access denied>
	Kernel driver in use: pcieport
	Kernel modules: shpchp
```

</p>
</details>

# Bit of explanation
scan_script.sh goes through the list of PCI devices.
When it encounters the [12d8:e110] device it then expects to find 4 [12d8:400a] devices right after the [12d8:e110]. When it happens it is assumed to be F223.
However in configuration where F213 is not detected there is [1172:d203] device right after [12d8:e110], so it is not considered to be F223 even that there are 4 [12d8:400a] devices right after that [1172:d203].
We think this approach is wrong, but comments in script say that this is the way it should be.

dpfeuffer, 2019-11-19:
> We think this approach is wrong, but comments in script say that this is the way it should be.

According to your lspci output, I agree with you that the approach is wrong.
Possible workarounds for F223 detection:
1) Skip all non "PCI bridge [0604]: Pericom Semiconductor" lines
or
2) Skip all devices at the same bus, until bus number changes (here: skip all further bus=0e entries)

But I don't know if this works always. It may depend on the PCI card plugged between CPU and F223.

dpfeuffer, 2019-11-19:
@FlorianRauh
The described problem affects the scan_sysgtem.sh script only.

KonradJPE, 2020-01-02:
Fixed.

closed 2020-01-02 by KonradJPE