MEN-Mikro-Elektronik/13MD05-90 #162: scan_system.sh must enable PCI memory regions

dpfeuffer, 2019-11-19:
The PCI memory regions of PCI devices are sometimes not enabled. See issue #74 

To be able to dump the chameleon table of FPGAs (fpga_load) and to read the M-Module Ids (mm_ident), scan_system.sh have to enable the memory regions of MEN PCI devices (MEN FPGAs with chameleon table and FPGAs of M-Module carriers).
This could be done with setpci. E.g.:
```sudo setpci -s 09:00.0 COMMAND=0x3```

dpfeuffer, 2019-11-25:
I tested your commit bc5199c94fee7b7bcb2b66cbdd2d296ab33eedae .
It seems to work, but it overwrites other bits in the PCI COMMAND register:

```
[root@centos ~]# setpci -s 07:00.0 COMMAND
0100

[root@centos ~]# ~/mdis/13MD05-90/MDISforLinux/scan_system.sh /opt/menlinux/
...
Found d203 G204 board with 1 M-Module A24 with disabled mem
Enable memory via setpci cmd
...
[root@centos ~]# setpci -s 07:00.0 COMMAND
0003
```

In this case, it clears bit 8 (SERR# Enable).

Please fix it:
1. read current value from COMMAND register
2. do a bitwise or with 0x3
3. write the modified value back

The Linux function pci_enable_resources() https://lxr.missinglinkelectronics.com/linux/drivers/pci/setup-res.c#L455 do it in similar way.

KonradJPE, 2019-12-02:
> Please fix it:
> 
>    1. read current value from COMMAND register
>    2. do a bitwise or with 0x3
>    3. write the modified value back

Fixed

closed 2019-12-02 by KonradJPE