MEN-Mikro-Elektronik/13MD05-90 #163: INSTALL.sh aborts when I would like to skip source installation

dpfeuffer, 2019-11-22:
First run of INSTALL.sh:
```
...
checking if /usr/src/linux/ points to valid kernel headers...
*** error: please set a symlink /usr/src/linux to the headers of your current running kernel:
           e.g.   ln -s /usr/src/linux-headers.x.y.z /usr/src/linux
In case of warning: "Cannot use CONFIG_STACK_VALIDATION=y"
please install libelf-dev, libelf-devel or elfutils-libelf-devel
*** error: prerequisites are not met, see help
```
Then I created the /usr/src/linux link and run INSTALL.sh again:
```
Would you like to proceed, and install MDIS sources into /opt/menlinux? (y/n/q): n
*** Aborted by user.
```
I would assume, that when I answer with 'n', the script proceeds without installing sources again into /opt/menlinux (which makes no sense, because already done).
But INSTALL.sh aborts regardless if I answer with 'n' or 'q'.
Script shall only abort with 'q' but shall proceed with next operation (skipping source installation) with 'n'.

By the way, please don't forget to update the version string:
```Installing the MEN MDIS for Linux System Package 13MD05-90_02_00```
But maybe you have this task already at your release checklist.

dpfeuffer, 2019-11-22:
Ok, I was too fast with my reporting, the next question apparently gives me the opportunity to skip the source installation, but it doesn't work either.
```
Checking if /opt/menlinux exists... 
Directory /opt/menlinux already exists.
Overwrite existing files? (y/n/q):
```
I also quits the installation when I try to proceed installation just without overwriting the sources:
```
Overwrite existing files? (y/n/q): n
*** Aborted by user.
```

Please change this behavior in this way:

1. 
'''Would you like to proceed, and install MDIS sources into /opt/menlinux? (y/n/q):'''
Shorten this question:
'''Would you like to install MDIS sources into /opt/menlinux? (y/n/q):'''
Change behavior:
y : proceed (ok)
n : skip installation of sources and proceed with next step (currently it aborts)
q : abort (ok)

2. 
```Overwrite existing files? (y/n/q):```
Change behavior:
y : proceed (ok)
n : skip installation of sources and proceed with next step (currently it aborts)
q : abort (ok)

dpfeuffer, 2019-11-22:
Another ugliness:
```
Would you like to build MDIS modules with the configuration from the system scan?
 Note: This requires installed Linux kernel sources/headers at the system! (y/n/q): y
Getting Compiler/Linker settings from Linux Kernel Makefile
Getting Compiler/Linker settings from Linux Kernel Makefile
```
Please remove second 'Getting Compiler/Linker settings ...' line or change it into another message (I don't what is actually checked).

dpfeuffer, 2019-11-28:
System scan:
system.dsc and Makefile shall be stored in the current working dir from which I called INSTALL but not where INSTALL resides.
Example:
```
# mkdir scan
# cd scan
# ~/mdis/13MD05-90/INSTALL.sh
scanning ...
# ls
# ls ~/mdis/13MD05-90
.. Makefile system.dsc ..
```

KonradJPE, 2019-12-31:
Fixed

closed 2019-12-31 by KonradJPE