MEN-Mikro-Elektronik/13MD05-90 #167: fpga_load is not correctly working with mcb and mcb_pci driver

FlorianRauh, 2019-12-02:
Starting from Linux kernel 3.15, modules mcb and mcb_pci are mainline in kernel.
But unfortunately fpga_load tool and mcb driver are interfering each other.

Therefore a hint as short term solution in the output of the fpga_load should be displayed:
"In order to run fpga_load correctly make sure that mcb and mcb_pci drivers are not loaded!"

Long term solution could be that fpga_load is not interfering with mcb and mcb_pci driver.

KonradJPE, 2019-12-03:
> Therefore a hint as short term solution in the output of the fpga_load should be displayed:
> "In order to run fpga_load correctly under Linux make sure that mcb and mcb_pci drivers are not loaded!"

Added. Please find compiled binary in release-13MD05-90_02_01 branch.
Additionally LINUX mak_switch was added into default fpga_load build.

FlorianRauh, 2019-12-09:
Dear Konrad, meanwhile I know that the source of the error was not the mcb drivers. 
The FPGA card itself was in a reset state and therefore the FPGA chameleon table was not accessible anymore. This had nothing to do with the mcb drivers. I apologize for a this issue. Please revert to the old source code and close the issue. My fault sorry.

KonradJPE, 2019-12-09:
@FlorianRauh Does it have to be reverted before 02_01 release?

FlorianRauh, 2019-12-09:
It´s actually wrong but it´s a tiny issue which is not harming the result in a negative way.  You can leave it for 02_01. 
But for 02_02 revert it please.

KonradJPE, 2019-12-30:
Fixed

closed 2019-12-30 by KonradJPE