MEN-Mikro-Elektronik/13MD05-90 #169: default selection of internal tools in mdiswiz not according package desc xml

dpfeuffer, 2019-12-02:
CentOs 7.5 3.10.0-862.el7.x86_64
On branch release-13MD05-90_02_01
13MD05-90_02_00-390-g1a9f671

```
|--4--|--3--|--2--|--1--|--2--|--3--|--4--| 
 F205  F215        F23P        G501  G215
0:M99                         SSD
1:-
```

I created an MDIS configuration with scan_system.sh and opened this configuration with mdiswiz. The selection of internal tools in mdiswiz differs from the xml file swmodules list with internal="true" attribute:

[image]

```
        <swmodule internal="true">
            <name>smb2_eeprod2</name>
            <description>Tool to dump, erase and program EEPROD2 board informations to SMBus devices via the SMB2_API</description>
            <type>Driver Specific Tool</type>
            <makefilepath>SMB2/TOOLS/SMB2_EEPROD2/COM/program.mak</makefilepath>
        </swmodule>
        <swmodule internal="true">
            <name>smb2_test</name>
            <description>Test tool for SMB2 functionality</description>
            <type>Driver Specific Tool</type>
            <makefilepath>SMB2/TOOLS/SMB2_TEST/COM/program.mak</makefilepath>
        </swmodule>
        <swmodule internal="true">
            <name>smb2_bmc</name>
            <description>Tool to control BMC features e.g. on F75P CPU boards</description>
            <type>Driver Specific Tool</type>
            <makefilepath>SMB2/TOOLS/SMB2_BMC/COM/program.mak</makefilepath>
        </swmodule>
```

- smb2_eeprod2 and smb2_test are selected but shouldn't.
- smb2_bmc_ctrl isn't selected but should.

See also #106

KonradJPE, 2019-12-03:
@dpfeuffer Could you please provide system.dsc with Makefile generated by scan_system script?

KonradJPE, 2019-12-03:
>   smb2_eeprod2 and smb2_test are selected but shouldn't.
>   smb2_bmc_ctrl isn't selected but should.

It should work properly now. 

Additionally I have removed "smb2_bmc" from default build which is also marked as \<swmodule internal\>

dpfeuffer, 2019-12-03:
>@dpfeuffer Could you please provide system.dsc with Makefile generated by scan_system script?

Du you still need the files?

>It should work properly now.

Good!

I thought that the scan script uses the xml description to add the tools, but now I see the changes in your commit and that it is hard coded in scan_system.sh itself. 
Is it possible to get the information out of the .xml to reduce hard coded stuff in scan_system.sh? Redundant information may run out of sync if we add/change any tool. (General question, not to consider it for 02.01 release).

KonradJPE, 2019-12-03:
> Du you still need the files?

No. 
But if you have easy access to hardware please confirm that it works.

> I thought that the scan script uses the xml description to add the tools, but now I see the changes in your commit and that it is hard coded in scan_system.sh itself.
> Is it possible to get the [...]

scan_system.sh uses .xml files for M-Modules and IpCores. So it is almost done. 
There are only several tools that are hard coded. After release we will look at it.

dpfeuffer, 2019-12-03:
>But if you have easy access to hardware please confirm that it works.

I made a quick test and it works.

>scan_system.sh uses .xml files for M-Modules and IpCores. So it is almost done.
There are only several tools that are hard coded. After release we will look at it.

Ok. I will remove MS 02.01. Please let this issue still open for 02.02 release.

KonradJPE, 2020-01-09:
> scan_system.sh uses .xml files for M-Modules and IpCores. So it is almost done.
> There are only several tools that are hard coded. After release we will look at it.Fixed, removed 

Fixed. Removed hard coded *.mak files in scan_system script for tools/modules that are described in *.xml files.

closed 2020-01-09 by KonradJPE