MEN-Mikro-Elektronik/13MD05-90 #170: scan_system.sh: M99 not detected

dpfeuffer, 2019-12-03:
CentOs 7.5 3.10.0-862.el7.x86_64
On branch release-13MD05-90_02_01
13MD05-90_02_00-390-g1a9f671

|--4--|--3--|--2--|--1--|--2--|--3--|--4--| 
 F205  F215        F23P        G501  G215
0:M99                         SSD
1:-

The M99 M-Module at the F205 carrier is not detected from scan_system.sh.
(no "99" string at all in Makefile/system.dsc.)

Same result with CentOs 8 + G25A + G204+M99.

dpfeuffer, 2019-12-03:
Here is the output from scanning with CentOs 8 + G25A + G204+M99

```
root@centos g215_detect_3]# /opt/menlinux/scan_system.sh /opt/menlinux/
============================================================
MDIS System Scan - generate initial system.dsc / Makefile
============================================================

/opt/menlinux/
checking if I2C tools exists...
I2C tools exists: PASSED
checking if PCI utils exists...
PCI utils exists: PASSED
checking if /usr/src/linux/ points to valid kernel headers...
/usr/src/linux/ points to valid kernel headers: PASSED
In case of warning: "Cannot use CONFIG_STACK_VALIDATION=y"
please install libelf-dev, libelf-devel or elfutils-libelf-devel
Scanning for MEN ID EEProm.
scan SMBus: bus 0 SMB addr 0x57
,,
scan SMBus: bus 4 SMB addr 0x57
found matching CPU: G025
Found CPU: G025. Using SMB address 4 for SMB2 based drivers
Writing system.dsc cpu section: _WIZ_MODEL = G25A
Writing CPU SMB BBIS section.
creating CPU SMB driver section: _WIZ_MODEL = smb2_1, SM Bus nr. = 4 at DEVICE_SLOT = 0
add_device_smb_scan_list
creating CPU SMB driver section: _WIZ_MODEL = xm01bc_1, SM Bus nr. = 4 at DEVICE_SLOT = 1
add_device_smb_scan_list
fill smb devices scan list in system.dsc file
Check if all required memory regions are enabled
Found d203 G204 board with 1 M-Module A24 with disabled mem
Enable memory via setpci cmd
Scanning for MEN PCI devices: 
Found d203 G204 board with 1 M-Module A24
Found d203_a24 no. 1
Writing d203_a24_1 section to system.dsc 
creating Makefile...
finished.
```

KonradJPE, 2019-12-03:
Will this issue be assigned to the current 02_01 release?

dpfeuffer, 2019-12-03:
Currently not, because M99 is only useful for us and testing. I'm using the M99 for acceptance testing, but I will refer to this issue in the test case (this will not result to a failed acceptance test result).
All other specified M-Modules shall be detectable without problems. I did no further testing with other M-Modules, but your tests shall cover it.

KonradJPE, 2020-04-30:
**G204:**
There is a problem with detection m99 on G204 board, and it is OS independent. We are unable to read M-Module ID.
**F205**
On F205 m99 is detected, however it works properly only when it is plugged into module slot 1.

After powering on the system, we can see the difference on LED(s). When plugged into F205, LED(s) are gently highlighted. On G204 nothing like that can be observed.

dpfeuffer, 2020-05-04:
Can you load the m99 driver at G204 without an error?  
Normally, all LL drivers are checking the M-Module id, but in m99_latency.c the id is not checked per default.  
In 13m09906.xml, I found this hint in the ID_CHECK description: "Define wether M-Module ID-PROM is checked, don't enable for M99, works only once after power up"
How often is the M-Module id read?

KonradJPE, 2020-05-04:
> Can you load the m99 driver at G204 without an error?

I have to manually add m99 driver into system.dsc and .mak files into Makefile.
After that I was able to load driver, and successfully run m99_latency simp program. 

> How often is the M-Module id read?

M-Module id is read during scan once, and it works after power up on F205 (seems to work for several times)
On G204 we are unable to read M-Module id even once. 

We are planning to test m99 on F205 during our automated tests, so 'test case' should be passed. Should we investigate it further, or just add a comment in the Test Report?

dpfeuffer, 2020-05-04:
No further investigation necessary, because the M99 is just a test M-Module.
And I'm lucky, because I'm using the M99 on an F205 for acceptance test.

KonradJPE, 2020-05-04:
> And I'm lucky, because I'm using the M99 on an F205 for acceptance test.

Could you please describe exact hardware configuration (module slot, other boards)?
We will confirm that it is working as expected, and then I will close the issue.

dpfeuffer, 2020-05-04:
>Could you please describe exact hardware configuration (module slot, other boards)?

Please see inSTEP Milestone 13MD05-90_02_01\13MD05-90_SwDeliveryAcceptanceReport.docx

KonradJPE, 2020-05-04:
Tested on Centos 8.1, configuration as below: 

> see inSTEP Milestone 13MD05-90_02_01\13MD05-90_SwDeliveryAcceptanceReport.docx

m99 is detected, and simple program m99_latency works properly. 
Please keep in mind that  m99 M-Module ID-PROM can be read only once after power up, so it works on the first run of scan_system.sh script.

mlassota, 2020-11-10:
I've recreated your configuration, but my HW+SW configuration have no problems detecting M99.
HW difference is that I have F614 attached to F23P.
SW seems to be the same:
```
[men@localhost ~]$ cat /etc/os-release 
NAME="CentOS Linux"
VERSION="8 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Linux 8 (Core)"
...

[men@localhost ~]$ uname -a
Linux localhost.localdomain 4.18.0-80.el8.x86_64 #1 SMP Tue Jun 4 09:19:46 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

However this is CentOS 8.0 and it is not supported by 02_03 release.

dpfeuffer, 2020-11-11:
Thanks for testing. I close the issue and remark in my acceptance test that I have to add the M99 manually.

closed 2020-11-11 by dpfeuffer