MEN-Mikro-Elektronik/13MD05-90 #174: M11 issue

KonradJPE, 2019-12-19:
Example program:
`m11_port_veri m11_1`
 sometimes fails.

m11_port_veri  output: 
> Testing 2X8 mode 
> Testing PortC -> PortB
> *** Verify error 1: port B is 0x30 should be 0x20

**M11 issue has been assigned to a ticket with the unique ID number 2982.**

dpfeuffer, 2020-06-19:
This issue was assigned to 13MD05-90_02_02. Is it solved?

dpfeuffer, 2020-06-19:
MAIN_PR006382 "04M011-02 ID check" belongs to this issue:

```
When the Module is mounted on a 02F205-00 the module verification will fail.

men@men:/usr/local/bin$ sudo m_open m11_1
open m11_1
*** can't open: ERROR (MDIS) 0x0505:  LL: wrong device id detected

After disabeling the ID check the tool will work again.

men@men:/usr/local/bin$ sudo m_open m11_1
open m11_1
path=3 opened
close path

After enabeling the debug function in mdis we were able to see in dmesg the same module shows different magic.

[   77.560194]  *** M11_Init: illegal magic=0x0100
[  222.651008]  *** M11_Init: illegal magic=0xf300

Testequipment:
Ubuntu 16.04
Linux men 4.10.0-28-generic #32~16.04.2-Ubuntu SMP Thu Jul 20 10:19:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
MDIS5 
02F021P05 Rev. 02.00.00
02F205-00
04M011-02 Rev.02.03.00
04M011-02 Rev.02.02.00
```

KonradJPE, 2020-06-19:
1. 

No, issue is still valid.
We can try to evaluate problem with `m11_port_veri`, but on the first look we haven't find anything that may cause this behaviour. 
If you would like us to look deeply into it, please move this to release 02_03. 

Do you have description for  **ID number 2982.** on your side?

2. 

For MAIN_PR006382 we have implemented workaround and disabled ID check in our tests.

dpfeuffer, 2020-06-19:
>Do you have description for ID number 2982. on your side?

I asked Martin. The description is just your e-mail conversation about this topic and the 2982 support ticket is closed.

Please investigate the problem. We should at least know the cause, then we can decide if it should be fixed.

mlassota, 2020-07-07:
I let the test run for 1.5h. The OK/ERROR ratio is 210/285.

Error statistics look like:
```
count | error
------+----------------------------------------------------
  219 | *** Verify error #1: port B is 0x00 should be 0x20
   25 | *** Verify error #6: port H is 0x02 should be 0x0
   16 | *** Verify error #6: port H is 0x08 should be 0x0
   25 | *** Verify error #10: port A is 0x00 should be 0x20
```
There's also a test (Test Hx signals) that makes memory leakage in kernel space:
```
unreferenced object 0xffff8f0e991d25a0 (size 16):
  comm "m11_port_veri", pid 8788, jiffies 4298143709 (age 2431.680s)
  hex dump (first 16 bytes):
    54 22 00 00 25 00 00 00 80 2c 05 87 0f 8f ff ff  T"..%....,......
  backtrace:
    [<00000000c0d9649e>] kmem_cache_alloc_trace+0x13a/0x250
    [<00000000b9394583>] men_OSS_SigCreate+0x48/0xd0 [men_oss]
    [<0000000091b66292>] M11_SetStat+0x41e/0x720 [men_ll_m11]
    [<000000004c26c309>] MDIS_LlSetStat+0x4f/0x80 [men_mdis_kernel]
    [<00000000a779a365>] mk_ioctl+0x409/0x1250 [men_mdis_kernel]
    [<00000000dc67cb2d>] mk_unl_ioctl+0xe/0x20 [men_mdis_kernel]
    [<0000000059ce1221>] do_vfs_ioctl+0xa9/0x640
    [<00000000f09a76d6>] ksys_ioctl+0x75/0x80
    [<000000005b164670>] __x64_sys_ioctl+0x1a/0x20
    [<0000000077f9f083>] do_syscall_64+0x5a/0x130
    [<000000002b7b3c3e>] entry_SYSCALL_64_after_hwframe+0x44/0xa9
unreferenced object 0xffff8f0e991d2df0 (size 16):
  comm "m11_port_veri", pid 8788, jiffies 4298143709 (age 2431.680s)
  hex dump (first 16 bytes):
    54 22 00 00 25 00 00 00 80 2c 05 87 0f 8f ff ff  T"..%....,......
  backtrace:
    [<00000000c0d9649e>] kmem_cache_alloc_trace+0x13a/0x250
    [<00000000b9394583>] men_OSS_SigCreate+0x48/0xd0 [men_oss]
    [<0000000091b66292>] M11_SetStat+0x41e/0x720 [men_ll_m11]
    [<000000004c26c309>] MDIS_LlSetStat+0x4f/0x80 [men_mdis_kernel]
    [<00000000a779a365>] mk_ioctl+0x409/0x1250 [men_mdis_kernel]
    [<00000000dc67cb2d>] mk_unl_ioctl+0xe/0x20 [men_mdis_kernel]
    [<0000000059ce1221>] do_vfs_ioctl+0xa9/0x640
    [<00000000f09a76d6>] ksys_ioctl+0x75/0x80
    [<000000005b164670>] __x64_sys_ioctl+0x1a/0x20
    [<0000000077f9f083>] do_syscall_64+0x5a/0x130
    [<000000002b7b3c3e>] entry_SYSCALL_64_after_hwframe+0x44/0xa9
unreferenced object 0xffff8f0e991d2a60 (size 16):
  comm "m11_port_veri", pid 8788, jiffies 4298143709 (age 2431.684s)
  hex dump (first 16 bytes):
    54 22 00 00 25 00 00 00 80 2c 05 87 0f 8f ff ff  T"..%....,......
  backtrace:
    [<00000000c0d9649e>] kmem_cache_alloc_trace+0x13a/0x250
    [<00000000b9394583>] men_OSS_SigCreate+0x48/0xd0 [men_oss]
    [<0000000091b66292>] M11_SetStat+0x41e/0x720 [men_ll_m11]
    [<000000004c26c309>] MDIS_LlSetStat+0x4f/0x80 [men_mdis_kernel]
    [<00000000a779a365>] mk_ioctl+0x409/0x1250 [men_mdis_kernel]
    [<00000000dc67cb2d>] mk_unl_ioctl+0xe/0x20 [men_mdis_kernel]
    [<0000000059ce1221>] do_vfs_ioctl+0xa9/0x640
    [<00000000f09a76d6>] ksys_ioctl+0x75/0x80
    [<000000005b164670>] __x64_sys_ioctl+0x1a/0x20
    [<0000000077f9f083>] do_syscall_64+0x5a/0x130
    [<000000002b7b3c3e>] entry_SYSCALL_64_after_hwframe+0x44/0xa9
```

dpfeuffer, 2020-07-08:
>Error statistics look like:

Please try to find the cause, because M11 is not EOL/EOS.
Maybe it is a HW problem of your M11 or a timing problem because of fast CPUs today (you could try to slow it down with some sleeps).

>There's also a test (Test Hx signals) that makes memory leakage in kernel space:

I guess that it comes from missing XXX_SIGCLR_XXX calls from application to driver (program mistake, or error handling problem).
Please investigate the M11 tools for SIGSET/SIGCLR pairs and consider in the driver's cleanup function to free all outstanding signal allocation with OSS_SigRemove.

mlassota, 2020-07-08:
> Maybe it is a HW problem of your M11 or a timing problem because of fast CPUs today (you could try to slow it down with some sleeps).

I added a 1 second delay after each M_* function call in test program and run it (only the part responssible for #1 error), but it dd not help.
Then I put M11 onto G214 and I plug G214 to G25A and did the test again. It worked. Then removed delays and it worked too. Then I did all the tests and again it worked.
It seems that it does not work on Test setup 2 (according to 13MD05-90_02_02-System-TestReport.docx).
And it seems to be a HW problem.

dpfeuffer, 2020-07-08:
Strange, I was thinking of a malfunctioning M11, but it works at G204+G25A according your test.
Test setup 2: F23P + CPCI backplane
Please make a test with test setup 5: F23P + hybrid backplane:
a) M11 at F205
b) M11 at G204

mlassota, 2020-07-08:
> Please make a test with test setup 5: F23P + hybrid backplane:
> a) M11 at F205
It doesn't work. 
Moreover when M11 is in slot 1 there is some PCI system error, because F205 appears several times in system:
```
$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller (rev 06)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06)
00:01.1 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x8 Controller (rev 06)
00:01.2 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x4 Controller (rev 06)
00:02.0 VGA compatible controller: Intel Corporation Haswell Integrated Graphics Controller (rev 06)
00:03.0 Audio device: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller (rev 06)
00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)
00:16.0 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection (2) I218-LM (rev 05)
00:1a.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 (rev d5)
00:1c.4 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 (rev d5)
00:1d.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation QM87 Express LPC Controller (rev 05)
00:1f.3 SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 05)
01:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
02:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
03:00.0 PCI bridge: Pericom Semiconductor PI7C9X110 PCI Express to PCI bridge (rev 04)
04:00.0 Bridge: Altera Corporation Device d203 (rev e3)
04:01.0 Bridge: Altera Corporation Device d203 (rev e3)
04:02.0 Bridge: Altera Corporation Device d203 (rev e3)
04:03.0 Bridge: Altera Corporation Device d203 (rev e3)
04:04.0 Bridge: Altera Corporation Device d203 (rev e3)
04:05.0 Bridge: Altera Corporation Device d203 (rev e3)
04:06.0 Bridge: Altera Corporation Device d203 (rev e3)
04:07.0 Bridge: Altera Corporation Device d203 (rev e3)
04:08.0 Bridge: Altera Corporation Device d203 (rev e3)
04:09.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0a.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0b.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0c.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0d.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0e.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0f.0 Bridge: Altera Corporation Device d203 (rev e3)
04:10.0 Bridge: Altera Corporation Device d203 (rev e3)
04:11.0 Bridge: Altera Corporation Device d203 (rev e3)
04:12.0 Bridge: Altera Corporation Device d203 (rev e3)
04:13.0 Bridge: Altera Corporation Device d203 (rev e3)
04:14.0 Bridge: Altera Corporation Device d203 (rev e3)
04:15.0 Bridge: Altera Corporation Device d203 (rev e3)
04:16.0 Bridge: Altera Corporation Device d203 (rev e3)
04:17.0 Bridge: Altera Corporation Device d203 (rev e3)
04:18.0 Bridge: Altera Corporation Device d203 (rev e3)
04:19.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1a.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1b.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1c.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1d.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1e.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1f.0 Bridge: Altera Corporation Device d203 (rev e3)
05:00.0 Bridge: Altera Corporation Device 203d (rev 04)
06:00.0 Bridge: MEN Mikro Elektronik Multifunction IP core

```

> b) M11 at G204

It works.

dpfeuffer, 2020-07-09:
to a)
So M11 in slot 0 at F205 produces the same M11 errors, but not the PCI system errors?
The PCI system errors may be caused from incorrect plugged F205 or damaged backplane/F205 pins.
But I fail to see why PCI errors shall depends on which slot the M11 is plugged.
Please double check it.

to b)
No problems with G204.

Please test with another F205. Maybe your F205 is malfunctioning?

mlassota, 2020-07-09:
> to a)
So M11 in slot 0 at F205 produces the same M11 errors, but not the PCI system errors?
The PCI system errors may be caused from incorrect plugged F205 or damaged backplane/F205 pins.
But I fail to see why PCI errors shall depends on which slot the M11 is plugged.
Please double check it.

Checked on other F205 and there was no PCI system error. So I assume I hadn't push it enough so it made no proper electric contact. But still ```*** Verify error #1: port B is 0x30 should be 0x20``` appears.

> to b)
No problems with G204.

Yep. So it has to be something with F205.

dpfeuffer, 2020-07-10:
Then we have to investigate this problem at MEN, because it seems to be a HW (F205 or M11) malfunction.
Please provide me the complete article/serial number of the M11/F205 you used for testing.

dpfeuffer, 2020-07-10:
Please give me also information of the used Linux distro and kernel version.

dpfeuffer, 2020-07-10:
Ok, i found the following info in your 13MD05-90_02_02-System-TestReport.pdf:
04M011-02  M11 Mezzanine Card with 16-bit TTL I/O Interface  02.02.00  006362  2 

But please provide info which of the F205 boards you used for testing.
Please test it with at last two F205 M-Module carriers.
Was the second M-Module slot empty?
Please make also a test with one M11 exclusively at F205.

dpfeuffer, 2020-07-10:
I created the MEN problem report MAIN_PR006869 for this issue.

mlassota, 2020-07-13:
> But please provide info which of the F205 boards you used for testing.

02F205-01 00.11.00 006815
02F205-00 00.11.00 006168

> Please test it with at last two F205 M-Module carriers.

Already did. Both F205's are affected.

> Was the second M-Module slot empty?

Already did separate tests for second M-Module empty and second M-Module occupied. No difference.

> Please make also a test with one M11 exclusively at F205.

Already did. On both F205s it makes no difference and makes no difference which slot is occupied.

> Please give me also information of the used Linux distro and kernel version.

Ubuntu 20.04 LTS
5.4.0-26-generic

> And used G204 M-Module carrier.

02G204-00 00.02.00 000006

dpfeuffer, 2020-07-13:
Thanks for the input.
I close this issue, as it seems to be impossible to solve in SW.
The problem will be investigated in MEN problem report MAIN_PR006869

closed 2020-07-13 by dpfeuffer