MEN-Mikro-Elektronik/13MD05-90 #175: BL51E CAN is not working

KonradJPE, 2019-12-19:
For HW rev. 01.01.00 CAN is not working. 

**Problem number is: MAIN_PR004890**

dpfeuffer, 2020-06-19:
MAIN_PR004890 "02F026L_pcie_boards_not_working_and_weird_behavior" describes:
```
OS: Linux, VxWorks

IP cores on G215 card not working:
- CAN
- UART
tested only under Linux: WIFI card on G-carrier also not working

F215 card OK.

something IRQ-related?

BIOS 1.05

weird things seen:
- PXE Boot in UEFI mode very slow (loading the very first files)
- PXE Boot in legacy mode Ubuntu 16.04.3 live the interface that was active during PXE cannot function in Linux later.
- USB Boot UEFI/legacy all interfaces are OK
- PXE Boot in UEFI mode all interface are OK

CPU	FPGA	OS	tool		result	
sc24	sc24	Lin	mscan_loopb	OK	
sc31	sc31	Lin	mscan_loopb	fail	
f26l	g215	Lin	mscan_loopb	fail	
sc31	sc31	Win10	mscan_loopb	OK	
f26l	g215	Win10	mscan_loopb	fail	
f22p	g215	Lin	mscan_loopb	OK	
f26l	g504 w nvme	Win10		fail	code 10 - device cannot start

MSCAN:

cat /proc/interrupts
            CPU0       CPU1       CPU2       CPU3
.
.
.
  21:          0          0          0          0   IO-APIC   21-fasteoi   can_1

at the same time:
sudo mscan_loopb can_1
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===

*** Error during: mscan_read_msg( path, rxObj, 1000, &rxFrm ) == 0
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 429
ERROR (MDIS) 0x0802:  OSS: timeout occurred
Test a: FAILED
=== Performing test b: Tx chronological                            (Run 1/1) ===

*** Error during: mscan_read_msg( path, rxObj, 1000, &rxFrm ) == 0
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 489
ERROR (MDIS) 0x0802:  OSS: timeout occurred
DRIVER INTERNALS:
$Id: mscan_drv.c,v 1.19 2013/03/27 09:54:36 gvarlet Exp $
MSCAN REGS:
 CTL0=00 CTL1=a0
 RFLG=00 TFLG=07 TIER=07
MSCAN DRIVER:
 txPrio: -1 -1 -1
MESSAGE OBJECTS:
 OBJ 0: rx
  totEntries: 10 filled: 0
 OBJ 1: tx
  txbUsed: 0 txNxtPrio 0 txSentPrio 15
  totEntries: 100 filled: 100
 OBJ 2: rx
  totEntries: 100 filled: 0

Test b: FAILED
=== Performing test c: Rx filter                                   (Run 1/1) ===

*** Error during: mscan_write_msg( path, txObj, 1000, &txFrm ) == 0
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 623
ERROR (MDIS) 0x0802:  OSS: timeout occurred
Test c: FAILED
=== Performing test d: Rx/Tx signals                               (Run 1/1) ===

*** Error during: entries == 10
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 780
ERROR (MSCAN) 0x0c0e:  illegal message object direction
Test d: FAILED
=== Performing test e: Rx FIFO overrun                             (Run 1/1) ===

*** Error during: entries == 10
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 844
ERROR (MSCAN) 0x0c0e:  illegal message object direction
Test e: FAILED
------------------------------------------------
TEST RESULT: 5 errors
Max irqtime=0 (internal ticks)
```

Check if this issue is related to  #196 and #199

dpfeuffer, 2020-06-19:
This issue was assigned to 13MD05-90_02_02. Is it solved?

KonradJPE, 2020-06-19:
This issue is hardware issue - for BL51E.
Some time ago Florian wrote about it. 
As I remember correctly UART (Rear interface) has also hardware bug.

dpfeuffer, 2020-06-19:
Thanks for input, I found this e-mail from 2019-12-06:

```
Dear Konrad,

With this HW Revision it is not possible so the Problem number is: MAIN_PR004890 is still vaild.
The next HW Revision or model with F1 stepping in Apollo Lake doesn´t have this error anymore.

Best Regards
Florian Rauh
```

I will clarify it with Florian.

dpfeuffer, 2020-06-19:
We will send you a new 09BL51E10 (F1 Stepping).
Please repeat the CAN testing with the new HW.

mlassota, 2020-07-08:
> We will send you a new 09BL51E10 (F1 Stepping).
> Please repeat the CAN testing with the new HW.

CAN works with the new hardware.

closed 2020-07-08 by mlassota