MEN-Mikro-Elektronik/13MD05-90 #177: License/Copyright - logos.h (16Z044)

KonradJPE, 2020-01-08:
What should be the correct license and copyright for file:
https://github.com/MEN-Mikro-Elektronik/13Z044-90/blob/jpe-dev/DRIVERS/FB_16Z044/TOOLS/Z44_256X64_TEST/logos.h ?

dpfeuffer, 2020-01-08:
Difficult question, because it contains not only a MEN logo but a Siemens logo. It's not only a license  question but also affects branding rights. It may take some time to clarify this.

dpfeuffer, 2020-02-14:
Please proceed as follows:
1) Replace MEN and Siemens logo in logos.h with two other images ("T E S T" and "Hello!").
2) Replace "sie_logo"/"Siemens logo" and "men_logo"/"MEN logo" in logos.h and fb16z044_256x64_test.c so that it fit to the two new images.
3) Add our standard copyright and GPL note to logos.h

To 1) and 2):
According to fb16z044_256x64_test.c, the display is a "256 * 64 * 4 monochrome displays".
Try to convert one of the char arrays from logos.h (MEN or Siemens logo) into a bitmap (.bmp file) that can be displayed with any graphic tool. Then create a new image (e.g. "T E S T") and try to convert it back into an array.
I know, you have no HW with this display to test the new images. Either we could test it somewhere or tbd.
If you think it is to complicated or time consuming, please let us know.
Other solution is to remove remove the logos but not replace it with new images.

mlassota, 2020-02-27:
Done. Please check if it's working.

JSierl, 2020-05-27:
Still needs Verification on our side. We will consider this issue "closed" regarding the milestone fulfillment.

dpfeuffer, 2022-11-18:
I just assume it is working and take the risk. It's not worth to spend time on testing currently because the test logo is nearly not used.

closed 2022-11-18 by dpfeuffer