MEN-Mikro-Elektronik/13MD05-90 #179: M65N not correctly detected in scan_system.sh

KonradJPE, 2020-01-13:
Add proper detection of M65N in scan_system.sh.
Differences between scan_system.sh and mdiswiz (system.dsc file): 
scan_system.sh:

- m65_x is added. 

mdiswiz:

- m65_xa and m65_xb with SUBDEVICE_OFFSET=0x00/0x80 parameter is added

dpfeuffer, 2020-01-14:
To 2.
Is the M65 plugged stand alone on an M-Module carrier? There is an errata about GND problems:
20m065-er.pdf (I added the document to InStep Customer\User Manuals).
Excerpt:
1.2.1 Description
There  are  ground  bounce  effects  on  the  connection  between  M-Module  and  carrier
board because of insufficient GND pins.
1.2.2 Projected Impact
In rare cases, errors occur when an another M-Module is plugged next to an M65 M-
Module on the same F205 carrier board.
1.2.3 Workarounds
Plug the M65 M-Module on a single-mezzanine carrier board, e.g., F204.

KonradJPE, 2020-01-14:
> 1.2.3 Workarounds
> Plug the M65 M-Module on a single-mezzanine carrier board, e.g., F204.

M65 M-Module is plugged into G204.

KonradJPE, 2020-01-27:
1. Fixed,
2. Moved to issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/181

closed 2020-01-27 by KonradJPE