MEN-Mikro-Elektronik/13MD05-90 #181: Error in M-Module M65 driver.

KonradJPE, 2020-01-27:
While performing icanl2_veri there are errors in 2 tests:
j - Remote transmit frames
k - Cyclic Tx/Timestamping

Error occurs when 
` ICANL2API_WriteObject `
or 
` ICANL2API_SendCyclic `
 function is used. The sent CAN frame does not contain valid data.

KonradJPE, 2020-01-28:
In code (icanl2_drv.c +533) there are comments like:
```
/* issue ICANL2_CMD_SEND command */ 
/* issue ICANL2_CMD_SEND_CYC command */
```
While using above commands CAN frames has invalid data.
Based on the comments it seems to be a known bug. Don't you already have a PR number for this?

Parameters reqData and reqLen passed to Command function are wrong.
To see the changes please look into commit. Now CAN frames are correct.

FlorianRauh, 2020-02-03:
Great.  Thank you Konrad. I agree to close this issue.

dpfeuffer, 2020-02-07:
```
/* issue ICANL2_CMD_SEND command */ 
/* issue ICANL2_CMD_SEND_CYC command */
```
" issue" is here not used in the meaning of "problem"
the comment means:
send/distribute/output XXX command

I take a look at your commit but I can't evaluate it on the quick without looking deeper into it.  But if you tested it and the problem's fixed, it should fit. Good job!

KonradJPE, 2020-02-10:
> I take a look at your commit but I can't evaluate it on the quick without looking deeper into it.

Ok. If you look at it, please get the newest sources from the M65 (jpe-dev) submodule. There is one more commit.

dpfeuffer, 2020-02-10:
Yes, I saw your commit, but I will not take the time to investigate it deeply. I just trust your test ;-)

KonradJPE, 2020-02-12:
Tests are passing correctly, so I'm closing the issue.

closed 2020-02-12 by KonradJPE