MEN-Mikro-Elektronik/13MD05-90 #182: fgpa_load aborts with segmentation fault while displaying chameleon table accessed by LPC/ISA bus

mlassota, 2020-01-28:
On BL70:
```
men@men:~$ sudo /opt/menlinux/BIN/fpga_load -k 0xe000e000 -t
Segmentation fault
```

mlassota, 2020-01-28:
fgpa_load still aborts with segmentation fault while displaying chameleon table accessed by LPC/ISA bus, so there's still some other bug.

```
men@men:~$ sudo /opt/menlinux/BIN/fpga_load -k 0xe000e000 -t

Chameleon FPGA table for device 0xe000e000:
  BAR0: 0x00000000; size: 0x00001000, mapType: IO;
  BAR1: 0xe0000000; size: 0x00010000, mapType: MEM;
  BAR2: 0x00000000; size: 0x00000000, mapType: unused;
  BAR3: 0x00000000; size: 0x00000000, mapType: unused;
  BAR4: 0x00000000; size: 0x00000000, mapType: unused;
  BAR5: 0x00000000; size: 0x00000000, mapType: unused;

Information about the Chameleon FPGA:
FPGA File='SC24-02IC026' table model=0x41('A') Revision 2.0 Magic 0xABCE
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x007d 16Z125_UART                0    4   0  17   6   0 0x00000220 0x220
  1 0x007d 16Z125_UART                0    0   0  17   4   0 0x000003f8 0x3f8
  2 0x007d 16Z125_UART                0    1   0  17   3   0 0x000002f8 0x2f8
  3 0x007d 16Z125_UART                0    2   0  17   7   0 0x000003e8 0x3e8
  4 0x007d 16Z125_UART                0    3   0  17   5   0 0x000002e8 0x2e8
  5 0x0052 16Z082_IMPULSE             0    0   0   2  63   1 0x00000800 0xe0000800
  6 0x0022 16Z034_GPIO                0    0   0  10  11   1 0x0000e200 0xe000e200
  7 0x0022 16Z034_GPIO                0    2   0  10  11   1 0x0000e240 0xe000e240
  8 0x0025 16Z037_GPIO                0    0   1   1  11   1 0x0000e260 0xe000e260
  9 0x007e 16Z126_SERFLASH            0    0   1   6  63   1 0x0000e280 0xe000e280
Segmentation fault
```

mlassota, 2020-01-28:
```
men@men:~$ sudo gdb -c core ./fpga_load
...
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0804f292 in Get_Chameleon (osHdl=0x9151f70, pciDev=0x0, tblAddr=0xe000e000, mod_id=45, chamInfo=0x91520e4, chamUnit=0x9152138, flags=1 '\001', pciDomain=0, mod_group=0) at /opt/menlinux/TOOLS/FPGA_LOAD/fpga_load.c:1842

warning: Source file is more recent than executable.
1842    z126Status = ((u_int32*)pRegs)[ (chamUnit->offset + SFII_BSR) >> 2 ];
(gdb) bt
#0  0x0804f292 in Get_Chameleon (osHdl=0x9151f70, pciDev=0x0, tblAddr=0xe000e000, mod_id=45, chamInfo=0x91520e4, chamUnit=0x9152138, flags=1 '\001', pciDomain=0, mod_group=0) at /opt/menlinux/TOOLS/FPGA_LOAD/fpga_load.c:1842
#1  0x0805088e in Z100_IsaInit (argv=0xffb78a54, argc=4, flashInterf=0, h=0x91520b8, osHdl=0x9151f70) at /opt/menlinux/TOOLS/FPGA_LOAD/fpga_load.c:2726
#2  main (argc=4, argv=0xffb78a54) at /opt/menlinux/TOOLS/FPGA_LOAD/fpga_load.c:525
```

KonradJPE, 2020-02-12:
Fixed

closed 2020-02-12 by KonradJPE