MEN-Mikro-Elektronik/13MD05-90 #183: M57+F205 creates PCI system problems

mlassota, 2020-02-10:
When M57+F205 is inserted into system it create some loop in PCI system making it showing 31 duplicates of this board:
```
$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller (rev 06)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06)
00:01.1 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x8 Controller (rev 06)
00:01.2 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x4 Controller (rev 06)
00:02.0 VGA compatible controller: Intel Corporation Haswell Integrated Graphics Controller (rev 06)
00:03.0 Audio device: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller (rev 06)
00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)
00:16.0 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection (2) I218-LM (rev 05)
00:1a.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 (rev d5)
00:1c.2 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 (rev d5)
00:1c.3 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 (rev d5)
00:1c.4 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 (rev d5)
00:1d.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation QM87 Express LPC Controller (rev 05)
00:1f.3 SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 05)
01:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
02:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
03:00.0 PCI bridge: Pericom Semiconductor PI7C9X110 PCI Express to PCI bridge (rev 04)
04:00.0 Bridge: Altera Corporation Device d203 (rev e3)
04:01.0 Bridge: Altera Corporation Device d203 (rev e3)
04:02.0 Bridge: Altera Corporation Device d203 (rev e3)
04:03.0 Bridge: Altera Corporation Device d203 (rev e3)
04:04.0 Bridge: Altera Corporation Device d203 (rev e3)
04:05.0 Bridge: Altera Corporation Device d203 (rev e3)
04:06.0 Bridge: Altera Corporation Device d203 (rev e3)
04:07.0 Bridge: Altera Corporation Device d203 (rev e3)
04:08.0 Bridge: Altera Corporation Device d203 (rev e3)
04:09.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0a.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0b.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0c.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0d.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0e.0 Bridge: Altera Corporation Device d203 (rev e3)
04:0f.0 Bridge: Altera Corporation Device d203 (rev e3)
04:10.0 Bridge: Altera Corporation Device d203 (rev e3)
04:11.0 Bridge: Altera Corporation Device d203 (rev e3)
04:12.0 Bridge: Altera Corporation Device d203 (rev e3)
04:13.0 Bridge: Altera Corporation Device d203 (rev e3)
04:14.0 Bridge: Altera Corporation Device d203 (rev e3)
04:15.0 Bridge: Altera Corporation Device d203 (rev e3)
04:16.0 Bridge: Altera Corporation Device d203 (rev e3)
04:17.0 Bridge: Altera Corporation Device d203 (rev e3)
04:18.0 Bridge: Altera Corporation Device d203 (rev e3)
04:19.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1a.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1b.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1c.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1d.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1e.0 Bridge: Altera Corporation Device d203 (rev e3)
04:1f.0 Bridge: Altera Corporation Device d203 (rev e3)
05:00.0 Bridge: Altera Corporation Device 203d (rev 04)
06:00.0 Bridge: Altera Corporation Device 203d (rev 04)
07:00.0 Communication controller: MEN Mikro Elektronik Multifunction IP core (rev 01)
08:00.0 Bridge: MEN Mikro Elektronik Multifunction IP core
```

dpfeuffer, 2020-02-10:
Please check if F205 is plugged properly and that F205 and backplane connectors are not damaged.

mlassota, 2020-02-10:
@dpfeuffer 
You're absolutely right. The board was not fully inserted.

closed 2020-02-10 by mlassota