MEN-Mikro-Elektronik/13MD05-90 #188: Fall through cases that seem to be invalid

mlassota, 2020-03-04:
Fall through in this cases seem to be invalid. Could you please verify if they really are invalid and just miss 'break' statement or this is related to #186 ?

[z51_drv.c +560](https://github.com/MEN-Mikro-Elektronik/13Z051-06/blob/jpe-dev/DRIVERS/MDIS_LL/Z051/DRIVER/COM/z51_drv.c#L560)
```/* select powerdown mode */
switch( value ) {
        case 1:  pdMode = DAC_CMD_PD_1K;
        case 2:  pdMode = DAC_CMD_PD_100K;
        case 3:  pdMode = DAC_CMD_PD_HIGHZ;
        default: pdMode = 0;
}
```
[z51_drv.c +568](https://github.com/MEN-Mikro-Elektronik/13Z051-06/blob/jpe-dev/DRIVERS/MDIS_LL/Z051/DRIVER/COM/z51_drv.c#L268)
```/* dependant on channel turn off output A or B */
switch( ch ) {
        case 0:
                MWRITE_D32( ma, DAC_CTRL_REG,
                        DAC_CMD_LOAD_A | DAC_CMD_BUF_A | pdMode );

        case 1:
                MWRITE_D32( ma, DAC_CTRL_REG,
                        DAC_CMD_LOAD_B | DAC_CMD_BUF_B | pdMode );
}
```

dpfeuffer, 2020-03-05:
Obvious two bugs:
- z51_drv.c +560: I can't see any sense of this switch/case constructions without any break. pdMode is always '0'.
- z51_drv.c +568: Also a bug.

Because I found also errors in the documentation, I made the following pull request to fix bugs and docu:
https://github.com/MEN-Mikro-Elektronik/13Z051-06/pull/1

mlassota, 2020-03-05:
Fix merged.

closed 2020-03-05 by mlassota