MEN-Mikro-Elektronik/13MD05-90 #191: INSTALL.sh: Failing creation of chameleon BBIS device

kaybouche, 2020-03-17:
Hi,

I'm having blocking issues with INSTALL.sh. I have 3 F405 cards installed and it seems, this is making the installer fail during system_scan.sh (even though it doesn't acutally fail, it just continues to run, but the output makes me think it should have aborted). The log is probably most expressive, I inserted "set -x" and "set +x" around the critical section. The first invocation of the function in question runs fine, the second one fails:

```
root@f26-bode:/home/men/13MD05-90# ./INSTALL.sh 
                                                                                
Installing the MEN MDIS for Linux System Package 13MD05-90_02_01                
(see MDIS User Manual for details)                                              
________________________________________________________________________________
                                                                                
Would you like to install MDIS sources into /opt/menlinux? (y/n/q): y
Checking if /opt/menlinux exists... 
Directory /opt/menlinux already exists.
Overwrite existing files? (y/n/q): n
Do not overwrite, proceed
Would you like to create an MDIS configuration for your system by scanning your system's hardware?
 Note: This make only sense at your MEN target system! (y/n/q): y
scanning system. Calling /opt/menlinux/scan_system.sh /opt/menlinux
============================================================
MDIS System Scan - generate initial system.dsc / Makefile
============================================================

/opt/menlinux
checking if I2C tools exists...
I2C tools exists: PASSED
checking if PCI utils exists...
PCI utils exists: PASSED
checking if /usr/src/linux/ points to valid kernel headers...
/usr/src/linux/ points to valid kernel headers: PASSED
In case of warning: "Cannot use CONFIG_STACK_VALIDATION=y"
please install libelf-dev, libelf-devel or elfutils-libelf-devel
backing up system.dsc...
Scanning for MEN ID EEProm.
scan SMBus: bus 0 SMB addr 0x57
found matching CPU: F026
Found CPU: F026. Using SMB address 0 for SMB2 based drivers
Writing system.dsc cpu section: _WIZ_MODEL = F26L
Writing CPU SMB BBIS section.
creating CPU SMB driver section: _WIZ_MODEL = smb2_1, SM Bus nr. = 0 at DEVICE_SLOT = 0
add_device_smb_scan_list
creating CPU SMB driver section: _WIZ_MODEL = xm01bc_1, SM Bus nr. = 0 at DEVICE_SLOT = 1
add_device_smb_scan_list
fill smb devices scan list in system.dsc file
Check if all required memory regions are enabled
Found F2xx/G2xx carrier with disabled memory region(s)
Enabling memory via setpci cmd:
Found F2xx/G2xx carrier with disabled memory region(s)
Enabling memory via setpci cmd:
Found F2xx/G2xx carrier with disabled memory region(s)
Enabling memory via setpci cmd:
Found F2xx/G2xx carrier with disabled memory region(s)
Enabling memory via setpci cmd:
Scanning for MEN PCI devices: 
Found possible MEN chameleon device(s), checking.
Building IP core database..........................done!
create chameleon BBIS device - based on lspci data
+++ lspci -d 1a88:4d45 -m
+++ grep 006a
+++ cut -f 1 -d ' '
++ lspci -s 0e:09.0 -v
+ local 'lspci_device_verbose_data=0e:09.0 Bridge: MEN Mikro Elektronik Multifunction IP core (rev 01)
	Subsystem: Device 006a:5a14
	Flags: 66MHz, medium devsel, IRQ 255
	Memory at 91630000 (32-bit, non-prefetchable) [size=4K]
	I/O ports at e000 [size=512]'
+ debug_args '0e:09.0 Bridge: MEN Mikro Elektronik Multifunction IP core (rev 01)
	Subsystem: Device 006a:5a14
	Flags: 66MHz, medium devsel, IRQ 255
	Memory at 91630000 (32-bit, non-prefetchable) [size=4K]
	I/O ports at e000 [size=512]'
+ '[' 0 == 2 ']'
++ echo '0e:09.0 Bridge: MEN Mikro Elektronik Multifunction IP core (rev 01)
	Subsystem: Device 006a:5a14
	Flags: 66MHz, medium devsel, IRQ 255
	Memory at 91630000 (32-bit, non-prefetchable) [size=4K]
	I/O ports at e000 [size=512]'
++ grep 'Capabilities.*Express Legacy Endpoint'
+ local is_pcie=
+ '[' -z '' ']'
+ echo 'Device: 1a88:4d45 PCI subvendor: 006a is CPCI device'
Device: 1a88:4d45 PCI subvendor: 006a is CPCI device
+ local pcibus_slot=2
+ local bus_if=cpu,1
++ expr 2 + 1
+ G_cPciRackSlotStandard=3
+ set +x
Writing sram_1 section to system.dsc
Writing uart_2 section to system.dsc
Writing uart_3 section to system.dsc
Writing rstctrl_4 section to system.dsc
Writing gpio_5 section to system.dsc
Writing gpio_6 section to system.dsc
Writing can_7 section to system.dsc
Writing can_8 section to system.dsc
Found possible MEN chameleon device(s), checking.
create chameleon BBIS device - based on lspci data
+++ lspci -d 1a88:4d45 -m
+++ grep 00d6
+++ cut -f 1 -d ' '
++ lspci -s 0e:0b.0 0e:0d.0 0e:0f.0 -v
Usage: lspci [<switches>]

Basic display modes:
-mm		Produce machine-readable output (single -m for an obsolete format)
-t		Show bus tree

Display options:
-v		Be verbose (-vv for very verbose)
-k		Show kernel drivers handling each device
-x		Show hex-dump of the standard part of the config space
-xxx		Show hex-dump of the whole config space (dangerous; root only)
-xxxx		Show hex-dump of the 4096-byte extended config space (root only)
-b		Bus-centric view (addresses and IRQ's as seen by the bus)
-D		Always show domain numbers

Resolving of device ID's to names:
-n		Show numeric ID's
-nn		Show both textual and numeric ID's (names & numbers)
-q		Query the PCI ID database for unknown ID's via DNS
-qq		As above, but re-query locally cached entries
-Q		Query the PCI ID database for all ID's via DNS

Selection of devices:
-s [[[[<domain>]:]<bus>]:][<slot>][.[<func>]]	Show only devices in selected slots
-d [<vendor>]:[<device>][:<class>]		Show only devices with specified ID's

Other options:
-i <file>	Use specified ID database instead of /usr/share/misc/pci.ids.gz
-p <file>	Look up kernel modules in a given file instead of default modules.pcimap
-M		Enable `bus mapping' mode (dangerous; root only)

PCI access options:
-A <method>	Use the specified PCI access method (see `-A help' for a list)
-O <par>=<val>	Set PCI access parameter (see `-O help' for a list)
-G		Enable PCI access debugging
-H <mode>	Use direct hardware access (<mode> = 1 or 2)
-F <file>	Read PCI configuration dump from a given file
+ local lspci_device_verbose_data=
+ debug_args ''
+ '[' 0 == 2 ']'
++ echo ''
++ grep 'Capabilities.*Express Legacy Endpoint'
+ local is_pcie=
+ '[' -z '' ']'
+ echo 'Device: 1a88:4d45 PCI subvendor: 00d6 is CPCI device'
Device: 1a88:4d45 PCI subvendor: 00d6 is CPCI device
+ local pcibus_slot=3
+ local bus_if=cpu,1
++ expr 3 + 1
+ G_cPciRackSlotStandard=4
+ set +x
Writing sram_9 section to system.dsc
^C
```

Regards
Kay

dpfeuffer, 2020-03-17:
Could you please investigate this issue the next days and give us a time estimation to fix it?

KonradJPE, 2020-03-17:
Which MDIS sources where used (branch)?

kaybouche, 2020-03-17:
I'm using the latest release from our website

KonradJPE, 2020-03-17:
Could you please check if the issue persist with sources cloned from 'jpe-dev' branch?

kaybouche, 2020-03-17:
Yes, it does. `lspci` still complains.
I would expect the entire process to fail as soon as `lspci` returns an error code, but the installer just continues.

kaybouche, 2020-03-17:
This is not a blocking issue anymore, I created my configuration manually.

KonradJPE, 2020-03-18:
The issue should be solved by the latest commit on jpe-dev branch.
Could you please run scan_system once again?

kaybouche, 2020-03-18:
Yes, it's working now. Thanks for fixing!

closed 2020-03-18 by kaybouche