MEN-Mikro-Elektronik/13MD05-90 #195: M99 on a203n A25 CPU is not working.

KonradJPE, 2020-05-28:
Configuration: 
- A203n plugged directly beneath the A25.  
- A203n rotary switch set to 0x000000
- M99 placed on slot 0/1/2 - the same results can be observed.

Modprobe drivers:
```
modprobe men_pldz002_cham
modprobe men_ll_m99_sw
```

No interrupt can be observed while running below commands:
**vme4l_m99irq:**
```
root@ubuntu:/home/ubuntu/a25# sudo vme4l_m99irq 0x0 0x80 3 1
vmeMap=0 m99Base=0x7ff282bda000
signal latency: Min: 2147483647, Max: 0 us, ints: 0
signal latency: Min: 2147483647, Max: 0 us, ints: 0
```
**m99_latency:**
```
root@ubuntu:~# sudo m99_latency m99_1
generating interrupts: timerval=250
(press any key for exit)
    current Interrupt-Latency        |     current Signal-Latency
  min[us]  avg[us]  max[us]  (irq/s) |  min[us]  avg[us]  max[us]  (sigs/s)
 =================================== | ====================================
     0   4294967295        0    (     0) |      0   4294967295        0    (     0)
4294967292   4294967295        0    (     0) | 4294967292   4294967295        0    (     0)
4294967292   4294967295        0    (     0) | 4294967292   4294967295        0    (     0)
4294967292   4294967295        0    (     0) | 4294967292   4294967295        0    (     0)
```
No dmesg error can be observed.
Interrupts with M-Modules: m35n, m77 works properly.

Attachements: 
[Makefile.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4695883/Makefile.txt)
[system.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4695888/system.txt)

dpfeuffer, 2020-06-19:
Please investigate that problem with low priority and take a look at VME4L_M99IRQ tool.

dpfeuffer, 2020-07-29:
Because the M99 is only a test M-Module, I remove the assigned 13MD05-90_02_03 milestone from this issue.

dpfeuffer, 2021-05-17:
- described in suspended MAIN_PR007544
- no further steps necessary because M99 is not intended for customer usage

closed 2021-05-17 by dpfeuffer