MEN-Mikro-Elektronik/13MD05-90 #196: BL50 CAN and interrupts problem

mlassota, 2020-06-01:
On BL50 *mscan_loopb* tool always reports an error while testing signals:
```=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===
Test a: ok
=== Performing test b: Tx chronological                            (Run 1/1) ===
DRIVER INTERNALS:
13Z015-06_02_26-2-g75f2524_2020-04-30
MSCAN REGS:
 CTL0=00 CTL1=a0
 RFLG=00 TFLG=07 TIER=00
MSCAN DRIVER:
 txPrio: -1 -1 -1 
MESSAGE OBJECTS:
 OBJ 0: rx
  totEntries: 10 filled: 0
 OBJ 1: tx
  txbUsed: 0 txNxtPrio 4 txSentPrio 3
  totEntries: 100 filled: 0
 OBJ 2: rx
  totEntries: 100 filled: 0

Test b: ok
=== Performing test c: Rx filter                                   (Run 1/1) ===
 Waiting transmit complete
 Verifying frames
Test c: ok
=== Performing test d: Rx/Tx signals                               (Run 1/1) ===

*** Error during: G_sigUos1Cnt == 10
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 768
ERROR (LINUX) #4:  Interrupted system call
Test d: FAILED
=== Performing test e: Rx FIFO overrun                             (Run 1/1) ===
Test e: ok
------------------------------------------------
TEST RESULT: 1 errors
Max irqtime=0 (internal ticks)
```

dpfeuffer, 2020-06-15:
Is the mscan_loopb test running at other HW without any problem?
Please investigate the problem.

mlassota, 2020-06-15:
There is also problem on F215 and G215, so I guess this is a mscan_loopb issue. Should we investigate it more?

dpfeuffer, 2020-06-16:
Yes please.

KonradJPE, 2020-06-17:
The problem is caused by the signals SIGUSR1, SIGUSR2 - which are no realtime.
After switching to realtime signals we cannot reproduce the issue. 
I will refer this issue in Test Report for release 02_02. 

Please find more details in issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/199

closed 2020-06-17 by KonradJPE