MEN-Mikro-Elektronik/13MD05-90 #197: BL51E SMBus problem on CentOS 7.7

mlassota, 2020-06-03:
On BL51E there is a problem of non working SMBus on CentOS 7.7:
i2cdetect log: [i2cdetect.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4724133/i2cdetect.log)
dmesg log: [dmesg.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4724132/dmesg.log)

There are these lines in dmesg log:
```
[   11.629163] i801_smbus 0000:00:1f.1: can't derive routing for PCI INT A
[   11.629170] i801_smbus 0000:00:1f.1: PCI INT A: no GSI
[   11.629247] i801_smbus 0000:00:1f.1: SMBus using PCI interrupt
```

And Anton pointed out this bug to be considered: [i2c: i801: Fix Failed to allocate irq -2147483648 error](https://patchwork.ozlabs.org/project/linux-i2c/patch/20171122112817.9554-1-hdegoede@redhat.com/)

dpfeuffer, 2020-10-07:
Are these problems happen with CentOS 7.7/7.8 only?
What about CentOS 8.2, Debian, Ubuntu?

dpfeuffer, 2020-10-07:
answer: see #211

closed 2020-11-13 by dpfeuffer