MEN-Mikro-Elektronik/13MD05-90 #198: EOS/EOL confusion

mlassota, 2020-06-05:
In the '13MD05-90_SwArchSpec' document, chapter 4.11 there is an information:

- a) '08SC24-' has the support state EOS
Does it mean that the BL50 should have this state too?
That would mean to add proper note in MDIS Wizard component list.

- b)  'ELinOS' has the support state EOL
Does it mean that 'ELinOS: Cross compilation...' option should be removed from MDIS Wizard development environment selection window?

dpfeuffer, 2020-06-15:
a) No, because BL50 shall be supported according MEN_13MD05-90_SA_1060.
But you are right, the BL50/SC24 requirements are confusing. I will clarify it with PM for 02.03.

b) No, but please add a EOL note if possible.

KonradJPE, 2020-06-15:
> b) No, but please add a EOL note if possible.

Should I add this EOL note to the current _02_02 release?

We are ready for the acceptance tests, and if it is possible I would like to postpone this change to the next release.

dpfeuffer, 2020-06-15:
It is not important for _02_02 release, you can postpone it.

KonradJPE, 2020-07-03:
> b) No, but please add a EOL note if possible.

I have added EOL note into "Select the development environment" window. @dpfeuffer Please take a look and confirm change or make a comment. 
[image]

dpfeuffer, 2020-07-03:
ok, thanks

closed 2020-07-03 by dpfeuffer