MEN-Mikro-Elektronik/13MD05-90 #199: Bug in USR_OSS library.

KonradJPE, 2020-06-17:
**1.** 

UOS_SIG_USR1 and UOS_SIG_USR2 signal shall be defined as realtime - as it is stated in file:  LIBSRC/USR_OSS/signal.c

```
  Realtime signals *will* be queued. (Totally max. 1024 for
  entire system in Linux 2.2). 
  So it's recommended to use only these signals. \c UOS_SIG_USR1 and
  \c UOS_SIG_USR2 are defined as realtime signals.
```

However signals defined in file INCLUDE/NATIVE/MEN/usr_os.h: 
``` 
/* some predefined signal codes */
# define UOS_SIG_USR1		(SIGUSR1)
# define UOS_SIG_USR2		(SIGUSR2)
```
are not realtime at all. This may cause unpredictable results while running MDIS tools - some signals might be missing (they are not queued). Please find more details in signal manual: 

```
Multiple instances of real-time signals can be queued.  By
contrast, if multiple instances of a standard signal are
delivered while that signal is currently blocked, then only one
instance is queued.
```

**2.** 

In file sja1000_test.c there are defines: 
``` 
# define UOS_SIG_USR3		(UOS_SIG_USR2+1)
# define UOS_SIG_USR4		(UOS_SIG_USR3+1)
```
Having in mind that UOS_SIG_USR2 is defined as SIGUSR2(12) leads to conclusion that USR3/USR4 signals refer to other OS signals SIGPIPE(13), SIGALRM(14) - which I suppose is not correct.

dpfeuffer, 2020-06-18:
Please check, if we can change in usr_os.h
```
# define UOS_SIG_USR1           (SIGUSR1)
# define UOS_SIG_USR2           (SIGUSR2)
```
into
```
# define UOS_SIG_USR1           (SIGRTMIN+1)
# define UOS_SIG_USR2           (SIGRTMIN+2)
# define UOS_SIG_USR3           (SIGRTMIN+3)
# define UOS_SIG_USR4           (SIGRTMIN+4)
```
Then we could leave sja1000_test.c unchanged.

Please think about my suggestion it an make some tests with some tools which are using signals.

mlassota, 2020-06-18:
The proposed solution would be perfect if it was a matter of kernel space, but it's a user space and therefore SIGRTMIN is defined as
```#define SIGRTMIN (__libc_current_sigrtmin ())```
so it actually is a function call, and since in MDIS tools and examples refere to UOS_SIG_USRn as case label (which has to be a literal, not a variable nor function)
```
sja1000_test.c-1082-static void SigHandler( u_int32 sigCode )
sja1000_test.c-1083-{
sja1000_test.c-1084-    switch(sigCode){
sja1000_test.c:1085:    case UOS_SIG_USR1:
sja1000_test.c-1086-            G_sigErr1++;
sja1000_test.c-1087-            break;
sja1000_test.c:1088:    case UOS_SIG_USR2:

```
the solutions are two:
**1. define RT signals as hard coded numbers**
```
# define UOS_SIG_USR1           (35)
# define UOS_SIG_USR2           (36)
# define UOS_SIG_USR3           (37)
# define UOS_SIG_USR4           (38)
```
This is actually against the manual, that discourages to use hard coded numbers to refer RT signals:
```
The Linux kernel supports a range of 33 different real-time signals, numbered 32 to 64.
However, the glibc POSIX threads implementation internally uses two (for NPTL)
or three (for LinuxThreads) real-time signals (see pthreads(7)), and adjusts
the value of SIGRTMIN suitably (to 34 or 35). Because the range of available
real-time signals varies according to the glibc threading implementation (and this
variation can occur at run time according to the available kernel and glibc),
and indeed the range of real-time signals varies across UNIX systems, programs
should never refer to real-time signals using hard-coded numbers, but instead
should always refer to real-time signals using the notation SIGRTMIN+n, and include
suitable (run-time) checks that SIGRTMIN+n does not exceed SIGRTMAX.
```
**2. do code refactoring to use *if* statement instead of *case* where appropriate**
At least 43 occurrences to refactorize:
```
men@men:~/13MD05-90 grep -rnH "case.*UOS_SIG_USR" --include="*.c" | wc -l
43
```

dpfeuffer, 2020-06-19:
Ok, then please implement solution 1 in the following way:
```
/*
 * We use linux real-time signals, because multiple instances of real-time signals
 * can be queued.
 * Warning: Becasue SIGRTMIN is not a fix value and defined as function call in
 * user space, it isn't usable in switch/case constructs. However, as workaround we
 * use here a fix value. On most linux distros SIGRTMIN is 34 or 35. 
 */
#define SIGRTMIN_FIX (35)
#define UOS_SIG_USR1 (SIGRTMIN_FIX+1)
#define UOS_SIG_USR2 (SIGRTMIN_FIX+2)
#define UOS_SIG_USR3 (SIGRTMIN_FIX+3)
#define UOS_SIG_USR4 (SIGRTMIN_FIX+4)
```

I found in a forum a signal problem that was fixed by using 36 instead of SIGRTMIN, because of that I suggest to use 36..39.

mlassota, 2020-06-22:
> I found in a forum a signal problem that was fixed by using 36 instead of SIGRTMIN, because of that I suggest to use 36..39.
I'd followed your suggestion.
Problem fixed.

closed 2020-06-22 by mlassota