MEN-Mikro-Elektronik/13MD05-90 #200: MDIS locking error causes kernel panic

dpfeuffer, 2020-06-18:
Problem is caused by issues in the locking mechanisms in MDIS Kernel, especially for the closing routine (mk_release). On frequent M_Open/M_GetStat/M_Close function calls it happens that after most of the device handles are already closed, different calls (BBIS/LL) are performed, either Write/SetStat/GetStat or even another closing cycle! This leads to accesses to already freed resources with kernel Oops and crashes.

Background: MDIS initializes the system 1x and then carries a useCount with it. Only with the close of the last Device (useCount==1) a de-initialization of the handle structures (SW) as well as the HW is really executed.

Problem is just as well reproducible with standard F26L (up to date BIOS/UEFI FW) and F215 (accessing the first Z034_GPIO for reading ("z17_io <dev> -g" cycles).
See [test_F215_Z34.zip](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4796725/test_F215_Z34.zip) for details.

Workaround:
Perform an "m_open <dev> -w &", for every device used, right after start of the system. Thereby the mk_release never calls the FinalClose routines and thereby the handles are never released/freed. This avoids the described behavior.

Work to do:
1. Reproduce the problem
2. Define a test setup to reproduce the problem. It would be good to use an already defined setup and create an test case that can be integrated into the regular MDIS release test.
3. Fix the problem
4. Make some stress tests

dpfeuffer, 2020-06-18:
Note for MEN: This issue belongs to MAIN_PR006749

KonradJPE, 2020-07-02:
>  1 . Reproduce the problem
>  3 . Fix the problem
>  4 . Make some stress tests

Done, after commit 415e127  we cannot reproduce the issue anymore. 

> 2 . Define a test setup to reproduce the problem. It would be good to use an already defined setup and create an test case that can be integrated into the regular MDIS release test.

For previous commits issue can be easly reproduced with test setups: 1,4
Would you like to create stress test cases for all simp tools/interfaces like CAN,UART,GPIO,ETH and M-Modules, or it will be enough to run this kind of test on GPIO only?

dpfeuffer, 2020-07-03:
It is enough to run the test on GPIO, because the locking issue was a general problem and is interface independent. But the test shall be executed before each MDIS release.

KonradJPE, 2020-07-09:
I have created automated stress test for z034_gpio.
We cannot observe kernel panic / locking error any more, so I'm closing this issue.

closed 2020-07-09 by KonradJPE