MEN-Mikro-Elektronik/13MD05-90 #202: vme4l_pldz002 driver doesn't start correctly with A21 FPGA A27 and older

dpfeuffer, 2020-06-19:
MAIN_PR006604 "13md05-90 vme4l_pldz002 driver doesn't start correctly with A21 FPGA A27 and older" describes this issue:

**Description**

```
When the vme4l-pldZ002-cham.ko module built from the latest versions of MDIS the driver can't start :

dmsg output : 

vme4l_pldz002_init_module 
vme4l_probe: probing 16Z002 unit
vme4l_probe: probing 16Z002 unit
vme4l_probe: probing 16Z002 unit
vme4l_probe: probing 16Z002 unit
!!!vme4l_pldz002_cham: vme4l_probe: Did not find PLDZ002 unit 7
!!!vme4l_pldz002_cham: vme4l_probe: Init error 22
vme4l_probe: probing 16Z002 unit
vme4l_probe: probing 16Z002 unit
vme4l_probe: probing 16Z002 unit

Trying a command on the VME controller :

# vme4l_ctrl sys
open /dev/vme4l_a24d16

*** Error during: (fd = VME4L_Open( VME4L_SPC_A24_D16 )) >= 0
file /opt/menlinux//TOOLS/VME4L_API/VME4L_CTRL/vme4l_ctrl.c
line 103
No such device or address
```

**Problem Analysis Result**

```
The problem comes from the numbers of Z002 units that the vme4l_pldz002_cham.c expects.
The number was increased to 8 to be usable for the A25, but i is then no longer usable on other VME boards : 

#define PLDZ002_MAX_UNITS	8

static int vme4l_probe( CHAMELEONV2_UNIT_T *chu )
{
...
	/* gather all the other chameleon units. Only A25 has 8 units, A21 and others 7 */
	for (i = 0; i < PLDZ002_MAX_UNITS ; i++)
	{
		if (men_chameleonV2_unit_find( CHAMELEON_16Z002_VME, i, &u) != 0) {
				printk(KERN_ERR_PFX "%s: Did not find PLDZ002 unit %d\n",
				       __func__, i);
			rv = -EINVAL;
			goto CLEANUP;
		}
...
```

**Project Impact/symptom**
```
No longer usable on A21 and older VME boards.
```

**Workaround**
```
Change the file MDISforLinux/DRIVERS/VME_16Z002/vme4l-pldz002-cham.c line 68

#define PLDZ002_MAX_UNITS	8

to

#define PLDZ002_MAX_UNITS	7
```

**Problem change suggestion / Measure**
```
The number of expected units should be dependant of the board the driver runs on.
```

mlassota, 2020-08-06:
Based on information provided by @dpfeuffer (i.e. Device ID: 0x4D45 and Subsystem Vendor ID: 0x00D5 are Coded for System: 16A025-00 [there is an error in User Manual for A25 stating Subsystem Vendor ID: 0x00D8 ]) I fixed the issue taking under consideration the mentioned IDs to identify A25.

mlassota, 2020-09-24:
I've tested my fix (9b909cc) on A25 and A21C and it works as expected.
Fixed.

closed 2020-09-24 by mlassota