MEN-Mikro-Elektronik/13MD05-90 #204: Global variable G_freeUsrBufList not protected for context switches / SMP

channoyer, 2020-07-03:
# Internal PR
MAIN_PR006854 F75P IOP - F215 frequently lose frame in multi-thread receive mode

# Background
This problem was observed with the [MSCAN driver](https://github.com/MEN-Mikro-Elektronik/13Z015-06) but can probably be reproduce with any other driver.

# Description of the problem
The application is executed on multi-core CPU. Two different threads access two different devices (can_1 and can_2).
In the application we observe frame lost (this is not suppose to happen because the two interfaces are connected on the same CAN bus and shall receive the same number of frames).

# Workaround
The user application shall protect all the mscan_xxx() api calls with semaphore.

# Correction
The correction applied in PR #203 fix the issue with our current test setup.

# Work to do
1. Reproduce the problem and define a test setup for this problem (if needed I can provide some test SW for this)
2. Check and confirm if the correction in PR #203 is enough
3. Make some stress tests

dpfeuffer, 2020-07-03:
Please consider to create a test case that will be executed before each MDIS release.

KonradJPE, 2020-07-06:
> Reproduce the problem and define a test setup for this problem (if needed I can provide some test SW for this)

@channoyer Please provide test SW if possible, it might speed up reproduction of the issue on our side.

channoyer, 2020-07-06:
Here is the current test app "as it is".
https://gist.github.com/channoyer/d8ecf3329c21e206c2fd426a3789eae1

Test setup:
DUT: F215 with two CAN interfaces.
In front of this one "Linux box" with one CAN interface running this:
`# while true ; do cangen can1 -g .2 -I 42A -e -L 5 -D i -n 20 ; sleep 0.04 ; done`

The 3 CAN devices are all connected on a single CAN bus.

KonradJPE, 2020-07-07:
> 1. Reproduce the problem and define a test setup for this problem (if needed I can provide some test SW for this)
> 2. Check and confirm if the correction in PR #203 is enough
> 3. Make some stress tests

Issue can be easly reproduced with F215 or G215 (Test Setup 1 for release 13MD05-90_02_03). I have performed stress test before and after PR. It seems that PR fix the issue.

Before the patch number of frames received on both interfaces differ, after patching I could not observe that behaviour.
However it rarely happens that the same packets were lost on both CAN interfaces (~3 packets per 1 000 000), for me it might not be releated to MDIS software. Should it be further examined?

@dpfeuffer 
> Please consider to create a test case that will be executed before each MDIS release.

I will add proper test case for release 02_03.

dpfeuffer, 2020-07-08:
>Before the patch number of frames received on both interfaces differ, after patching I could not observe that behaviour.
However it rarely happens that the same packets were lost on both CAN interfaces (~3 packets per 1 000 000), for me it might not be releated to MDIS software. Should it be further examined?

@channoyer 
What do you think? Please decide it.

>I will add proper test case for release 02_03.

Thanks

channoyer, 2020-07-08:
Hum I do not expect packet loss at all. How did you detect this packet loss?
Why do you think it's not related to MDIS software? What is for you the origin of these packet loss?

channoyer, 2020-07-13:
@KonradJPE I want to ask you if you try to understand the packet loss on your side?
I ask this because on our side it seems our initial problem is not completely resolved.

KonradJPE, 2020-07-20:
First of all sorry for my late reply, I was on holidays. 
Please find my answers below: 

> Hum I do not expect packet loss at all. How did you detect this packet loss?

I have compared the number of packets send on can0 interface(external USB-CAN adapter), and the packets received on MDIS can(s). On MDIS side, I have used your test program (10 packets are missing).

> Why do you think it's not related to MDIS software? What is for you the origin of these packet loss?

I though it migh be problem related to physical connection between interfaces, but after verification and performing more tests it seems to be MDIS SW/HW issue. Please find below connection scheme.

[image]

Please take a look into attached log from your test program. 
[mscan_concurrent.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/4946383/mscan_concurrent.log)

> @KonradJPE I want to ask you if you try to understand the packet loss on your side?
> I ask this because on our side it seems our initial problem is not completely resolved.

We will take a look into MDIS code as it is described in #205. Hope we will find something. 
Yes, it seems that your initial problem still exists. 

If any hints/ideas come to your mind what we should first check, please give a comment.

channoyer, 2020-07-27:
Hi and sorry also for my late reply.

I just want to tell you that maybe the problem could be also related to the test SW itself.

Also FYI on the initial system we found this issue we are using preempt rt kernel. With this system for the moment we found a correct working configuration by setting all the priorities correctly according to our needs.

For your test are you also using preempt rt kernel?

Maybe you can create kernel traces when the loos of frames appear and check if the CAN interrupt handler was not preempted for a too long period?

KonradJPE, 2020-07-28:
> I just want to tell you that maybe the problem could be also related to the test SW itself.

I have played a bit with your test program, and I could not reproduce the issue after reducing thread sleep time
`usleep(40000);`
Moreover I have run numerous 'cangen' loops with different options, but I could not observe any specific pattern that may help find the cause. 

> For your test are you also using preempt rt kernel?

No we do not use preempt rt kernel patch. 

> Maybe you can create kernel traces when the loos of frames appear and check if the CAN interrupt handler was not preempted for a too long period?

Unfortunately after adding logs it is almost impossible to see an error - it might run for hours.
Furthermore till now we have not found more suspicious code in MDIS kernel that may influence this CAN behavior.

> Also FYI on the initial system we found this issue we are using preempt rt kernel. With this system for the moment we found a correct working configuration by setting all the priorities correctly according to our needs.

So after setting priorities you cannot observe this issue on your side anymore?

@dpfeuffer, 
Do you encounter/heard of any other strange problems with other interfaces like UART,GPIO so we will be able to reproduce it?

channoyer, 2020-07-28:
> So after setting priorities you cannot observe this issue on your side anymore?

Yes.

dpfeuffer, 2020-07-29:
>Do you encounter/heard of any other strange problems with other interfaces like UART,GPIO so we will be able to reproduce it?

Currently not, I will forward your question to our support.

mlassota, 2020-08-03:
Seems there's no more work to do.

closed 2020-08-03 by mlassota

dpfeuffer, 2020-08-24:
> @dpfeuffer,
Do you encounter/heard of any other strange problems with other interfaces like UART,GPIO so we will be able to reproduce it?

No, neither I nor our support department have additional input.