MEN-Mikro-Elektronik/13MD05-90 #208: MDIS tools don't work with Ubuntu 20.04

dpfeuffer, 2020-09-09:
This issue refers to the MAIN_PR006953 problem report.

Detected in Revision
2.1

Initiating
Bernhard Höher

Description
Tools e.g. hwbug accessing /dev/mem may not work with Ubuntu 20.04

Problem Analysis Result
With Ubuntu 20.04 Kernel 5.4.0-45-generic the tools accessing /dev/mem may not work, because of CONFIG_STRICT_DEVMEM=y in Kernel forbidding access to /dev/mem even as root.

Can this problem be solved within MDIS or is it required that the customer builds a kernel with CONFIG_STRICT_DEVMEM=n?

mlassota, 2020-09-10:
@dpfeuffer Could you please provide the output of the hwbug when it's not working?

I've tried to recreate this problem on 'Ubuntu 20.04 LTS' with '5.4.0-26-generic' kernel release running on 'x86_64' CPU (this is a configuration I'm using for MDIS functional tests).
The only problem I've encountered is the lack of 32-bit libraries installed by default, and since hwbug_x86 is 32-bit dynamically linked application you won't be able to run it:
```
root@men:/opt/menlinux/TOOLS/HWBUG# ./hwbug_x86 
bash: ./hwbug_x86: No such file or directory
```
Thus you need to install 32-bit support libraries:
```
root@men:/opt/menlinux/TOOLS/HWBUG# dpkg --add-architecture i386
root@men:/opt/menlinux/TOOLS/HWBUG# apt-get update
root@men:/opt/menlinux/TOOLS/HWBUG# apt-get install libc6-i386
```
Then you can test it:
```
root@men:/opt/menlinux/TOOLS/HWBUG# ./hwbug_cmd_x86 -d -i 0x10 0x40

i/o mapped access

Dump memory 10..40
00000010+00000000: 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff 0f ................
00000010+00000010: 20 ff ff ff 20 ff ff ff 20 ff ff ff 20 ff aa 00  ... ... ... ...
00000010+00000020: 20 ff ff ff 20 ff ff ff 20 ff ff ff 20 ff ff ff  ... ... ... ...
```
So at this point we either add the note to HWBUG's README.md that it needs 32-bit libraries to run or we link hwbug statically (just like fpga_load is) so it was self-sufficient.
And I checked if kernel .config file has CONFIG_STRICT_DEVMEM=y turned on - yes, it has:
```
root@men:/opt/menlinux/TOOLS/HWBUG# grep -nH "DEVMEM" /usr/src/linux-headers-5.4.0-26-generic/.config 
/usr/src/linux-headers-5.4.0-26-generic/.config:4200:CONFIG_DEVMEM=y
/usr/src/linux-headers-5.4.0-26-generic/.config:10441:CONFIG_ARCH_HAS_DEVMEM_IS_ALLOWED=y
/usr/src/linux-headers-5.4.0-26-generic/.config:10442:CONFIG_STRICT_DEVMEM=y
/usr/src/linux-headers-5.4.0-26-generic/.config:10443:# CONFIG_IO_STRICT_DEVMEM is not set
```
Also read from '/dev/mem' is working:
```
root@men:/opt/menlinux/TOOLS/HWBUG# head -c 1056 /dev/mem | hexdump -C
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000400  00 00 00 00 00 00 00 00  00 00 00 00 00 00 c0 9e  |................|
00000410  00 00 00 7b 02 00 00 00  00 00 1e 00 1e 00 00 00  |...{............|
00000420
```

dpfeuffer, 2020-09-11:
> Could you please provide the output of the hwbug when it's not working?

I forwarded your question to my colleague who described this problem, but he is unavailable  till 15.09.

dpfeuffer, 2020-09-16:
Correction from our colleague:
"
The cause seems to be not CONFIG_STRICT_DEVMEM=y but the so-called "Linux Kernel Lock Down". This Lock down is coupled to Secure Boot. As soon as Secure Boot is activated, the kernel lock down is also active. The lock down implies a lockout of /dev/mem itself as root: https://mjg59.dreamwidth.org/50577.html I had indeed unintentionally activated secure boot. If I deactivate secure boot in bios, the error message "Operation not permitted" is gone. The lockdown feature was introduced with kernel 5.4 in April 2018 and is part of Ubuntu 20.04 and older Ubuntus, which have been updated accordingly: https://mjg59.dreamwidth.org/55105.html 
"

For next release 2021:
Please evaluate if it is possible to check for activated lockdown feature within INSTALL.sh and then show an appropriate warning message that /dev/mem is not accessible and hwbug and fpga_load not usable. This check/warning message should also be implemented in hwbug/fpga_load itself.

mlassota, 2020-12-14:
Checks and warning messages added.

closed 2020-12-14 by mlassota