MEN-Mikro-Elektronik/13MD05-90 #209: request_module() doesn't work on CentOS 8.2 due to SELinux restrictions making men_bbis_kernel unable to load needed drivers

mlassota, 2020-09-10:
```
[men@localhost ~]$ sudo modprobe men_ll_m34
[sudo] password for men: 
[men@localhost ~]$ dmesg
[   62.944087] men_dbg: loading out-of-tree module taints kernel.
[   62.944146] men_dbg: module verification failed: signature and/or required key missing - tainting kernel
[   62.944413] MEN men_dbg init_module
[   62.957262] MEN men_oss init_module
[   62.959161] MEN men_desc init_module
[   62.961736] MEN BBIS Kernel init_module
[   62.968705] MEN MDIS Kernel init_module
[   62.971828] MEN men_id init_module
[   62.974180] MEN men_mbuf init_module
[   62.976295] MEN men_ll_m34 init_module
[men@localhost ~]$ sudo m34_simp m35_1 14
[sudo] password for men: 

13M034-06_02_06-0-gf62987d_2019-11-26

*** ERROR (MDIS) 0x0b06:  BK: board handler not found ***
    => Error
[men@localhost ~]$ dmesg
[   68.908758]  try to load men_bb_d203_a24 trough kerneld
[   68.911748] *** BBIS:request_module: error: 256
[   68.911760] *** BBIS:OpenDevice: can't find driver men_bb_d203_a24
[   68.911761] OSS_Exit
[   68.911762] bbis_open ex: d203_a24_2 ret=-2822 -0xb06
[   68.911763] *** MK:InitialOpen: : m35_1 can't open BBIS dev d203_a24_2
[   68.911764] MK - FinalClose:
[   68.911764] MK - FinalClose: device=m35_1
[   68.911765]  irq installed: 0
[   68.911765]  close board
[   68.911766] bbis_close: d203_a24_2
[   68.911767] bbis_close ex: d203_a24_2 ret=-2 -0x2
[   68.911767] OSS_Exit
[   68.911768] *** MK:InitialOpen:  m35_1 failed error=0xb06
[   68.911778] *** MDIS_OpenDevice: m35_1: initial open failederr=0xb06
[   68.911779] DESC - DESC_Exit
[   68.911780] mk_ioctl exit: cmd=0x40704d05 ret=-2822 (-0xb06)
[   68.911785] mk_release file=000000002ab3a37a
```

One way to solve the problem is to edit file '/etc/selinux/config' and replace the line `SELINUX=enforcing` with `SELINUX=permissive`, then reboot OS.
Whether there's an other way requires further investigation.

dpfeuffer, 2020-09-11:
>Whether there's an other way requires further investigation.

Please take the time to investigate this problem, because CentOS 8.2 support is required for the 13MD05-90_02_03 release.
If I understand it right, SELINUX=enforcing is the default configuration and CentOS 8.2 default installation would leed to this problem. Am I right?

mlassota, 2020-09-11:
> If I understand it right, SELINUX=enforcing is the default configuration and CentOS 8.2 default installation would leed to this problem. Am I right?

Yes, you're right.

mlassota, 2020-09-11:
> Please take the time to investigate this problem, because CentOS 8.2 support is required for the 13MD05-90_02_03 release.

I don't see the way we could workaround this problem in MDIS. For security reasons one module cannot load other module directly when SELinux is in enforce mode.

dpfeuffer, 2020-09-16:
Decision:
@mlassota: Please check for ``SELINUX=enforcing`` configuration within the INSTALL.sh script and show a appropriate warning that the user has to set  ``SELINUX=permissive`` in '/etc/selinux/config'.

I will describe the necessity for this setting in the User Manual and Quickstart guide.

mlassota, 2020-10-09:
Fixed

closed 2020-10-09 by mlassota