MEN-Mikro-Elektronik/13MD05-90 #211: DC19 SMBus problem on CentOS 7.8

mlassota, 2020-10-06:
On DC19 there is a problem of non working SMBus on CentOS 7.8:
```
[men@localhost ~]$ sudo i2cdetect -l
i2c-0   i2c             i915 gmbus dpb                          I2C adapter
i2c-1   i2c             i915 gmbus dpc                          I2C adapter
i2c-2   i2c             i915 gmbus misc                         I2C adapter
i2c-3   i2c             DPDDC-B                                 I2C adapter
i2c-4   i2c             DPDDC-C                                 I2C adapter
i2c-5   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-6   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-7   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-8   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-9   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-10  i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-11  i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-12  i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-13  smbus           SMBus I801 adapter at f040              SMBus adapter
[men@localhost ~]$ sudo i2cdetect -y 13
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
[men@localhost ~]$ dmesg | tail
[ 1686.706898] i801_smbus 0000:00:1f.1: Timeout waiting for interrupt!
[ 1686.709287] i801_smbus 0000:00:1f.1: Transaction timeout
[ 1686.913922] i801_smbus 0000:00:1f.1: Timeout waiting for interrupt!
[ 1686.916312] i801_smbus 0000:00:1f.1: Transaction timeout
[ 1687.121028] i801_smbus 0000:00:1f.1: Timeout waiting for interrupt!
[ 1687.123426] i801_smbus 0000:00:1f.1: Transaction timeout
[ 1687.327940] i801_smbus 0000:00:1f.1: Timeout waiting for interrupt!
[ 1687.330333] i801_smbus 0000:00:1f.1: Transaction timeout
[ 1687.535039] i801_smbus 0000:00:1f.1: Timeout waiting for interrupt!
[ 1687.537431] i801_smbus 0000:00:1f.1: Transaction timeout
```
This can be related to "BL51E SMBus problem on CentOS 7.7 #197"

dpfeuffer, 2020-10-07:
Are these problems happen with CentOS 7.7/7.8 only?
What about CentOS 8.2, Debian, Ubuntu?

mlassota, 2020-10-07:
> Are these problems happen with CentOS 7.7/7.8 only?

Yes, it seems so. BL51E and DC19 are both SC31 machines.
However issue #197 was tested on 09BL51E00 on CentOS 7.7. Recently I repeated the test on 09BL51E10 (F1 Stepping) on CentOS 7.8 and there was not porblem.

> What about CentOS 8.2, Debian, Ubuntu?

These OS's are not affected.

dpfeuffer, 2020-10-07:
Please repeat tests for this issue and #197 with CentOS 7.7 and 7.8. Provide uname -a output (minor distro version, kernel version) and version of i2c-tools. Can be done after 13MD05-90_02_03 release. 
I think we have to provide an errata for this problem. Are there any known workaround?

mlassota, 2020-10-07:
>  Are there any known workaround?

I'm not aware of any.

closed 2020-11-13 by dpfeuffer