MEN-Mikro-Elektronik/13MD05-90 #212: DC19 shuts down after Power Button "being pressed" on Debian 10.5 32- and 64-bit

mlassota, 2020-10-06:
Except there's no power nor suspend button.
There is no problem on other tested OSes (Ubuntu 18.04.3 32-bit, Ubuntu 18.04.3 64-bit, Ubuntu 20.04 64-bit, CentOS 7.8 64-bit, CentOS 8.2 64-bit).

What helped:
1) Adding these lines to */etc/systemd/logind.conf*
```
HandlePowerKey=ignore
HandleSuspendKey=ignore
HandleLidSwitch=ignore
HandleLidSwitchExternalPower=ignore
PowerKeyIgnoreInhibited=yes
SuspendKeyIgnoreInhibited=yes
HibernateKeyIgnoreInhibited=yes
```
2) Appending kernel command line parameters in grub with ```3``` to stay in text mode.

However OS still reports key press event (but just ignores it):
```
men@men:/var/log$ sudo tail auth.log
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men systemd-logind[419]: Power key pressed.
Oct  6 15:44:47 men sudo:      men : TTY=pts/0 ; PWD=/var/log ; USER=root ; COMMAND=/usr/bin/tail auth.log
Oct  6 15:44:47 men sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
```

dpfeuffer, 2020-10-07:
Debian 10.5 is not required for 13MD05-90_02_03.
Could you please repeat the test with another Debian (e.g. the required 10.4). Maybe we talked about using 10.5 instead of 10.4? But I would like to know if all Debians are affected.

mlassota, 2020-10-07:
> Debian 10.5 is not required for 13MD05-90_02_03.
You are right. However I couldn't find 10.4 installer, so I used 10.5 (actually the network installer) and we agreed that it is even better to use more up to date Debian version.

> Could you please repeat the test with another Debian (e.g. the required 10.4). Maybe we talked about using 10.5 instead of 10.4?
I did the test with 10.6 (which is now most recent; network installer just installs most recent version) and it works fine.

dpfeuffer, 2020-10-07:
We will provide an errata for this Debian 10.5 specific problem.

dpfeuffer, 2020-10-08:
@mlassota 
Do you observed the same behavior on BL51E with on Debian 10.5 32- and 64-bit?

mlassota, 2020-10-08:
> Do you observed the same behavior on BL51E with on Debian 10.5 32- and 64-bit?

No.

closed 2020-11-13 by dpfeuffer

FlorianRauh, 2020-11-19:
The topic is unfortunately incomprehensible. Within Debian 10.5 a message appears from nowhere, that a Power Button "being pressed" which leads to a shutdown. Whereas there is no PowerButton.

dpfeuffer, 2020-11-27:
We can't reproduce the described behavior with Debian 10.5 64-bit.
Debian 10.5 32bit boot failed with error "The kernel does not support 64bit CPUs".