MEN-Mikro-Elektronik/13MD05-90 #213: DC19 serial console outputs garbage

mlassota, 2020-10-06:
I've tried to deal with issue #212 using serial console but it outputs garbage. The console redirection is default (I used "Load Defaults" in BIOS). Client configuration (minicom) matched those default settings. I've tried different settings (speed, parity etc.) but it didn't work.

dpfeuffer, 2020-10-07:
1) Am I right, that the problem concerns the BIOS redirection prior to Linux boot (no Linux involved)?
2) Does the same UART interface work under Linux?
3) Which BIOS version does your DC19 have?

mlassota, 2020-10-07:
> 1. Am I right, that the problem concerns the BIOS redirection prior to Linux boot (no Linux involved)?

Yes, that is right. This is not MDIS related problem.

> 2. Does the same UART interface work under Linux?

Yes. I did tests using loopback to verify this (but with Console Redirection disabled).

> 3. Which BIOS version does your DC19 have?

1ATJS 1.08 x64

dpfeuffer, 2020-10-07:
Please make an additional test under Linux with a connection to another system (not just a loopback test). According to F.R. we have had the same behavior (garbage output) caused by a HW defect. It could be possible that the DC19-->LOOPBACK-->DC19 seems ok, when the HW malfunction influences TX and RX in the same way.

mlassota, 2020-10-07:
I did test DC19-->RS232->MY_DESKTOP_PC (both running Linux) and the output is a mess (in both directions). So you're right, it's HW defect.

dpfeuffer, 2020-10-07:
Send us the DC19 for repair after 13MD05-90_02_03 tests are finished.

closed 2021-01-13 by mlassota

mlassota, 2021-01-14:
Turns out, that User Manual, chapter 2.7.6 Switching between RS232 and RS422/RS485 Interface, explains everything.