MEN-Mikro-Elektronik/13MD05-90 #214: SC31 - smb2 tools are not build

M-Gerner, 2020-10-22:
When installing MDIS5 on a system with a SC31 (BC51M, DC19) the smb2 tools (smb2_eetemp, etc.) were not build. The driver is installed sucessfully.

The customer was able to find a way to solve the issue for himself through modifying the makefile. They copied two lines from F011 to SC31 (see pictures).
[image]
[image]

dpfeuffer, 2020-10-23:
I think the MDISforLinux/BUILD/MDIS/TPL/DSC/sc31.tpl file adds SMB2 and XM01BC drivers, and the
```
add_mdis_drivers "XM01BC"
add_mdis_drivers "SMB2"
```
lines are used to add SMB2 and XM01BC tools.

M-Gerner, 2020-10-23:
I have double checked it with a SC24, SC31 and JPE Dev Branch.

On both (SC24 and SC31) the drivers (men_ll_smb2, men_ll_xm01bc) and xm01bc_ctrl were build. But smb2 Tools like smb2_eetemp are missing.
In our User Manuals we have examples with the smb2 Tools. It is inconsistent to show the customer the tools but they are not available to the customer.

dpfeuffer, 2020-10-23:
The cause of the problem is the MDISforLinux/BUILD/MDIS/TPL/DSC/Makefile.sc31.tpl file (and it seems to me, that other Makefile.sc*.tpl files are also affected). This makefiles add the SMB2 driver but not the SMB2 tools.
I'm wondering if the Makefile*.tpl files are still required. The information which SW to build should be taken from the driver package description files (xml).

dpfeuffer, 2020-10-23:
@mlassota 
Please check the described behavior and how to improve the scan functionality by using xml instead of Makefile*.tpl files.
All non internal SMB2 tools shall be available when the SMB2 driver will be installed.

mlassota, 2021-01-13:
Fixed missing tools.

dpfeuffer, 2021-02-25:
I reopen this issue, because we run today into the following problem with 13MD05-90_02_03:
When we add just the smb2 tools manually from the MDISWIZ, without adding the necessary smb2 libs, we get build errors.
IMHO is your commit to fix this issue are not complete, because it adds no SMB2 libs.
Please try to reproduce these build errors by adding the 
ALL_LL_TOOLS = \
from your commit (changes in Makefile.sc31.tpl).
Then add the missing SMB2 API libs (reported from the build errors) and fix it.

mlassota, 2021-03-03:
Fixed

closed 2021-03-03 by mlassota