MEN-Mikro-Elektronik/13MD05-90 #222: 13Z077-90 MAC from EEPROM appears invalid

inotb, 2021-02-09:
The Z77 driver states the (correctly programmed) MAC is invalid.

good:        13Z077-90_02_03-11-gddc3cd3_2020-02-25

current MAC 00:00:00:00:00:00 is invalid, try get one from an attached MAC EEPROM.
got MAC 00:c0:3a:d5:20:ba from MAC EEPROM, assigning it.

bad:        13Z077-90_02_04-0-ga63eac2_2020-11-17
bad:        13Z077-90_02_04-3-gc633fd9_2021-01-28

MAC 00:c0:3a:d5:00:ba from MAC EEPROM is invalid or no MAC EEPROM attached, try get one from Board ID EEPROM.
i2c_designware i2c_designware.0: controller timed out
i2c_designware i2c_designware.1: controller timed out
MAC from BOARD EEPROM is invalid or no board EEPROM found. Resorting to random MAC.

dpfeuffer, 2021-02-10:
Thanks for your report, but who are you and what HW product are you using?
Please report problems with our products via the official support request form https://www.duagon.com/support/contact/

inotb, 2021-02-10:
!!!! after a short investigation it turned out that the "good" driver is only more tolerant and the "bad" driver checks a bit at the end of the MAC which is aparity or CRC bit.
see conversation Z77 MAC invalid attached. an unknown number of F614 cards must have been programmed incorrectly. The driver might not need to be changed.

closed 2021-02-10 by inotb