MEN-Mikro-Elektronik/13MD05-90 #223: Automate functional testing with setup 7

dpfeuffer, 2021-04-26:
Q (dpfeuffer):
What is the reason that the functional testing with setup 7 (VMEbus rack with A25) was performed manually? Would it be possible to automate it like for the other test setups?

A (jpe):
I think that because of the scan_system.sh not detecting A203N and an extra steps need to be taken to create Makefile and system.dsc and we were in rush back then, we abandoned this idea and decided to come back to it later, but we didn't. It's definitely possible to run it automatically.

dpfeuffer, 2021-05-17:
Please consider this input to improve testing for the next release.

mad-jsanjuan, 2022-10-17:
I believe the [mad-dev-gonzalo](https://github.com/MEN-Mikro-Elektronik/mdis_test_system/compare/mad-dev-gonzalo) branch from the [mdis_test_system](https://github.com/MEN-Mikro-Elektronik/mdis_test_system) was trying to accomplish automation of test setup 7.

closed 2023-03-09 by mad-jsanjuan