MEN-Mikro-Elektronik/13MD05-90 #224: Improve testing

dpfeuffer, 2021-04-26:
Q (dpfeuffer):
Do you have any other suggestions to improve the testing? 

A (jpe):
The test system is complex. I would put configuration into one file and add parameter to specify configuration file. I'd use some simple relay to turn on and off the testing voltage instead of using photo-coupler in Box PC.

dpfeuffer, 2021-05-17:
Please consider this input to improve testing for the next release.

afernandez-tech, 2021-07-13:
Please, Can you explain me this changes? what is the number test?
thanks

dpfeuffer, 2021-07-13:
>what is the number test?

"number test"? I don't understand.

This issue address general improvements for the testing. Please investigate what can be improved. See also the remark from Marcin (jpe) above.

We can also talk about it at the next MDIS meeting.

dpfeuffer, 2022-11-18:
Jorge and Javier are now in progress to improve the testing. I close this issue because it contains no further helpful info.