MEN-Mikro-Elektronik/13MD05-90 #228: mscan_pingpong does not work

dpfeuffer, 2021-06-11:
We discovered the following problem with the MDIS MSCAN driver package https://github.com/MEN-Mikro-Elektronik/13Z015-06/tree/master/DRIVERS/MDIS_LL/MSCAN

Description from our Problem Report MAIN_PR006912:

There are multiple bugs in the mscan_pingpong example application:
 - It will throw:
   line 280
   ERROR (MSCAN) 0x0c12:  controller not disabled
   if can has been enabled in some other application before. mscan_pingpong should disable the CAN devices itself before trying to configure them.
 - It simply does not work and returns "OSS: timeout occured" when trying to receive the TX packets on the other interface. CAN interfaces, however, are working properly as confirmed with other test programs sending packets between them.

Test hardware: F215
Invocation: mscan_pingpong can_7 can_8
system.dsc is attached
[system.dsc.txt](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/6637024/system.dsc.txt)

Please investigate this problem.
If you need further support/input about this problem, I will give you the e-mail address from the colleague who discovered the problem.

GonzaloMartinR, 2021-10-20:
We believe we have found a solution to these, modifying the scheduling policy momentarily. 

[image]

With these changes we are consistently getting 0 errors regardless of the number of tests we run.

dpfeuffer, 2021-11-10:
Today we decided to postpone the MSCAN driver problem. The investigating requires deep knowledge of the driver and FPGA IP core functionality and is time consuming. We do not have the time to support you with this task at the moment. I lower down the priority and remove it from the 13MD05-90_02_05 milestone.