MEN-Mikro-Elektronik/13MD05-90 #229: mdis_drivers_bbis_chameleon_com bugfix

dpfeuffer, 2021-07-14:
Note: This issue belongs to the internal problem report MAIN_PR007117 "15EM10Axx QDEC Hänger"

The following pull request shall be merged and tested:
https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com/pull/3

This bugfix was necessary because of a customer problem under VxWorks.
The fix was tested under VxWorks only but should be also required/usable under Linux.
It is sufficient to perform the usual compilation/functional tests, there is no need for a special test to be carried out.

VictorParro, 2021-07-16:
Please Dieter, can you confirm if running a Test Setup 2 is enough to validate this modification

dpfeuffer, 2021-07-16:
It is sufficient when you run all the standard functional tests during the normal verification procedure of the next 13MD05-90_02_05 release.
Test setup 2 is not suitable to test the chameleon BBIS driver, because the driver is not used. The chameleon BBIS handles chameleon FPGAs (e.g. at F215/G215 boards) but not M-Module carriers.
My suggestion is to make a basic compilation test now. And a basic functional test with any test setup that uses the chameleon BBIS (if you have already one running). However, before the 02_05 release, all tests must be performed.

VictorParro, 2021-07-16:
Thanks Dieter. It wasn't clear to me, which setup the BBIS Cahmeleon driver is used. To validate this modification, I will follow your suggestion.

closed 2021-11-16 by GonzaloMartinR