MEN-Mikro-Elektronik/13MD05-90 #230: mdis_drivers_bbis_chameleon_com PCI_BUS_PATH bugfix

dpfeuffer, 2021-09-10:
Note: Belongs to internal e-mail thread "[Work Item 1245996028] - [ TC2130007 ] - 02F026L02_05_06_02F229_PCI_BUS_NUMBER_of_02F614_fpga_changes"
Tasks:
- Handle pull request https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com/pull/4
- Create new test case to check PCI_BUS_PATH functionality (see MDIS UG for details) with 2-3 CPU boards (I tested with F23P).
- The PCI_BUS_PATH is set in system.dsc when you create a manual MDIS configuration with the MDIS Wizard (e.g. specify CompactPCI slot number for mezz_cham carrier board F215). PCI_BUS_PATH depends on the used CPU board.
- The new test case should be part of the test setups for 13MD05-90 release test.

dpfeuffer, 2022-11-18:
Victor merged the pull request 2021-11-15.
New test case should be added for PCI_BUS_PATH verification (see above).

mad-jrodriguez, 2022-12-13:
I have added 2 Test Cases to verify the PCI_BUS_PATH feature for F26L and F27P CPU Boards.

A new Pull Request have been created to merge the test cases.

https://github.com/MEN-Mikro-Elektronik/mdis_test_system/pull/18.

The tests work as expected

closed 2022-12-16 by mad-jrodriguez