MEN-Mikro-Elektronik/13MD05-90 #240: z77 Frame Padding

M-Gerner, 2022-06-14:
During the test campaign of the STAC release where vulnerability tests for the F26L2 ethernet interfaces done.
Ethernet frames created by men_lx_z77 can contain padding-bytes which are different from frame to frame, which allows attacker to obtain information from previous packets or kernel memory by using malformed packets.

The CVE-ID https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2003-0001

Siemens is using 13MD05-90 version 2.3. According to change log of version 2.04 there were no changes at the z77 driver so it should be still there.

dpfeuffer, 2022-11-18:
[Z77 driver repo](https://github.com/MEN-Mikro-Elektronik/13Z077-90)

dpfeuffer, 2022-11-18:
This issue is the same as MAIN_PR008157