MEN-Mikro-Elektronik/13MD05-90 #242: Evaluate commits in mad-dev_02_05 and merge it into mad-dev

dpfeuffer, 2022-10-14:
According the Configuration management plan 13MD05-90_SwConfMgmtPlan.md, the development branch of the SCM project repos shall be named mad-dev.
There is a mad-dev_02_05 branch with commits from the retired Spanish colleagues. Please evaluate the commits and merge it into mad-dev. 
The development branch shall be used later on for the release testing.
All pull requests shall be merged into mad-dev, but not into master directly.
For further information, see chapter "Workflow C - Outsourced projects" in "Guideline for SW Configuration Management (SW-CM)".

mad-jsanjuan, 2022-10-17:
On a first investigation it looks the submodules intended to be updated for next release would be:

```
[submodule "DRIVERS/BBIS/CHAMELEON"]
	branch = mad-dev
[submodule "13Z015-06"]
	branch = mad-dev_issue228
[submodule "13M072-06"]
	branch = mad-dev
[submodule "13Z055-90"]
	branch = mad_dev
```

These branches are also inconsistent in the naming (mad_dev, mad-dev. mad-dev_*). An overview is described below:

# [CHAMELEON](https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com)

This repo contains one branch with changes that may be needed for next release:

 * [mad-dev](https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com/compare/mad-dev)
 
# [13Z015-06](https://github.com/MEN-Mikro-Elektronik/13Z015-06)

In this repo there are two mad* branches: 
 * [mad-dev_issue228](https://github.com/MEN-Mikro-Elektronik/13Z015-06/compare/mad-dev_issue228) : This one contains one commit that does not seem to be useful. It seems to add debugging traces.
 * [mad-dev-10_14](https://github.com/MEN-Mikro-Elektronik/13Z015-06/compare/mad-dev-10_14): This one contains some commits that may affect the drivers for this components.

# [13M072-06](https://github.com/MEN-Mikro-Elektronik/13Z072-06)

This repo contains one branch with changes that may be needed for next release:

 * [mad-dev](https://github.com/MEN-Mikro-Elektronik/13Z072-06/compare/mad-dev)

# [13Z055-90](https://github.com/MEN-Mikro-Elektronik/13Z055-90)

This repo contains one branch with changes that may be needed for next release. However, there seems to be some development based on `master` instead of this `mad_dev`branch. We should probably rename this branch to `mad-dev` for consistency and make sure expected developments for next releases are based on this branch instead of master. 

 * [mad_dev](https://github.com/MEN-Mikro-Elektronik/13Z055-90/compare/mad_dev) : This one shall be named `mad-dev` instead. 
 * [TC2219001](https://github.com/MEN-Mikro-Elektronik/13Z055-90/compare/master...ckloecker:13Z055-90:TC2219001) : This development may be needed for next release. If so it would need to be merged to `mad-dev`.

dpfeuffer, 2022-10-19:
A few comments to your investigation results:

# [CHAMELEON](https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com)

 * [mad-dev](https://github.com/MEN-Mikro-Elektronik/mdis_drivers_bbis_chameleon_com/compare/mad-dev)
These changes are from NBG (JFriesz and myself) and should be fine.

# [13Z015-06](https://github.com/MEN-Mikro-Elektronik/13Z015-06)

 * [mad-dev_issue228](https://github.com/MEN-Mikro-Elektronik/13Z015-06/compare/mad-dev_issue228) : useless branch
 * [mad-dev-10_14](https://github.com/MEN-Mikro-Elektronik/13Z015-06/compare/mad-dev-10_14): these changes are a try to fix the issue: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/228
We postponed these issue and I suggest to skip it further on. 

# [13M072-06](https://github.com/MEN-Mikro-Elektronik/13Z072-06)

 * [mad-dev](https://github.com/MEN-Mikro-Elektronik/13Z072-06/compare/mad-dev): ok, just cosmetics and compiler satisfaction

# [13Z055-90](https://github.com/MEN-Mikro-Elektronik/13Z055-90)

 * [mad_dev](https://github.com/MEN-Mikro-Elektronik/13Z055-90/compare/mad_dev)
 * [TC2219001](https://github.com/MEN-Mikro-Elektronik/13Z055-90/compare/master...ckloecker:13Z055-90:TC2219001) #243

The changes in both branches are to be adopted for the next version according to your suggestion.

closed 2022-12-15 by mad-jsanjuan