MEN-Mikro-Elektronik/13MD05-90 #246: .kernelsettings cannot get created with Debian 11 linux headers

mad-jsanjuan, 2022-11-08:
The kernel headers in debian 11 is split into two directories:

* linux-headers-5.10.0-19-amd64: Configured as entry point for building the kernel modules. Actual link in `/lib/modules/$(uname -r)/build/Makefile`
* linux-headers-5.10.0-19-common: This one contains the original Makefile and some common sources.

Make variables coming from the included `/usr/src/linux-headers-5.10.0-19-common/Makefile` do not get exported to the upper level `kernelsettings.mak` so the .kernelsettings file gets empty values.

This is a summary on how to reproduce this issue:

```
men@debian:~/Work/MDIS/mdis_test_system/MDIS_Compilation_Test$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

men@debian:~/Work/MDIS/mdis_test_system/MDIS_Compilation_Test$ ls -al /usr/src/
total 16
drwxr-xr-x  4 root root 4096 Nov  7 14:52 .
drwxr-xr-x 14 root root 4096 Nov  7 14:08 ..
drwxr-xr-x  4 root root 4096 Nov  8 15:39 linux-headers-5.10.0-19-amd64
drwxr-xr-x  4 root root 4096 Nov  8 15:39 linux-headers-5.10.0-19-common
lrwxrwxrwx  1 root root   24 Oct 21 21:24 linux-kbuild-5.10 -> ../lib/linux-kbuild-5.10

men@debian:~/Work/MDIS/mdis_test_system/MDIS_Compilation_Test$ cat /lib/modules/$(uname -r)/build/Makefile
include /usr/src/linux-headers-5.10.0-19-common/Makefile

men@debian:~/Work/MDIS/mdis_test_system/MDIS_Compilation_Test$ sudo make --trace -C /usr/src/linux-headers-5.10.0-19-amd64 --no-print-directory -f /media/tests/MDIS_Test/MDIS_Compilation/13MD05-90_Install/BUILD/MDIS/TPL/kernelsettings.mak KERNEL_SETTINGS_FILE=.kernelsettings
/media/tests/MDIS_Test/MDIS_Compilation/13MD05-90_Install/BUILD/MDIS/TPL/kernelsettings.mak:43: target 'getsettings_for_mdis' does not exist
echo "KERNEL_CC := " > .kernelsettings
echo "KERNEL_LD := " >> .kernelsettings
echo "KERNEL_CFLAGS := " >> .kernelsettings
echo "KERNEL_LDFLAGS := " >> .kernelsettings
echo "KERNEL_ARCH := " >> .kernelsettings
```

Here is an expample on how this works in Ubuntu:

```
men@men-F027:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
men@men-F027:~$ ls -al /usr/src/
total 24
drwxr-xr-x  6 root root 4096 nov  8 07:47 .
drwxr-xr-x 11 root root 4096 sep 15  2021 ..
lrwxrwxrwx  1 root root   40 nov  8 07:47 linux -> /usr/src/linux-headers-5.4.0-131-generic
drwxr-xr-x  7 root root 4096 nov  8 17:08 linux-headers-5.4.0-131-generic
drwxr-xr-x  7 root root 4096 sep 15  2021 linux-headers-5.4.0-84-generic
drwxr-xr-x 24 root root 4096 nov  7 09:16 linux-hwe-5.4-headers-5.4.0-131
drwxr-xr-x 24 root root 4096 sep 15  2021 linux-hwe-5.4-headers-5.4.0-84
men@men-F027:~$ 
men@men-F027:~$ sudo make --trace -C /usr/src/linux-headers-5.4.0-131-generic --no-print-directory -f /opt/menlinux/BUILD/MDIS/TPL/kernelsettings.mak KERNEL_SETTINGS_FILE=.kernelsettings
/opt/menlinux/BUILD/MDIS/TPL/kernelsettings.mak:38: target 'getsettings_for_mdis' does not exist
echo "KERNEL_CC := gcc" > .kernelsettings
echo "KERNEL_LD := ld" >> .kernelsettings
echo "KERNEL_CFLAGS := " >> .kernelsettings
echo "KERNEL_LDFLAGS := " >> .kernelsettings
echo "KERNEL_ARCH := x86" >> .kernelsettings
```

mad-jsanjuan, 2022-11-10:
I tried to repdroduce this issue on Debian 10 and got the following:

The linux headers package is also split into two directories.

```
$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

$ ls -al /usr/src/
total 16
drwxr-xr-x  4 root root 4096 Nov 10 11:56 .
drwxr-xr-x 14 root root 4096 Nov 10 11:15 ..
lrwxrwxrwx  1 root root   38 Nov 10 11:56 linux -> /usr/src/linux-headers-4.19.0-22-amd64
drwxr-xr-x  4 root root 4096 Nov 10 11:21 linux-headers-4.19.0-22-amd64
drwxr-xr-x  4 root root 4096 Nov 10 12:28 linux-headers-4.19.0-22-common
lrwxrwxrwx  1 root root   24 Sep 29 02:47 linux-kbuild-4.19 -> ../lib/linux-kbuild-4.19
```

The `.kernelsettings` created gets some values but these are not the one that should have been extracted from the kernel's Makefile. It seems it reads CC and LD from the make default environment:

mad-jsanjuan, 2022-11-11:
Another note from the investigation. We could have Debian 11 work in the same way it does in Debian 10 by letting make get environment variables with `-e` option. This would not be the initial intention of `.kernelsettings` as it should read those from the Kernel's Makefile but may help debugging this issue.

From a Debian 11 run:
```
$ sudo make -e -C /usr/src/linux-headers-5.10.0-19-amd64 --no-print-directory -f /media/tests/MDIS_Test/MDIS_Compilation/13MD05-90_Install/BUILD/MDIS/TPL/kernelsettings.mak KERNEL_SETTINGS_FILE=.kernelsettings
echo "KERNEL_CC := cc" > .kernelsettings
echo "KERNEL_LD := ld" >> .kernelsettings
echo "KERNEL_CFLAGS := " >> .kernelsettings
echo "KERNEL_LDFLAGS := " >> .kernelsettings
echo "KERNEL_ARCH := " >> .kernelsettings
```

dpfeuffer, 2022-11-18:
see also #249

closed 2022-12-15 by mad-jsanjuan